particle-iot/device-os #993: digitalRead() interferes with analogRead()

ScruffR, 2016-05-05:
https://community.particle.io/t/an-input-ok-to-use/22660/2

Based on my understanding of the implementation of `analogRead()` and confirmed by Brett in [this post](https://community.particle.io/t/solved-analog-pin-configuration-outside-of-setup/20147/13) I suggested to a member that it is possible to set a `pinMode(analogPin, INPUT_PULLUP)` in order to use `digitalRead(analogPin)` and a subsequent call to `analogRead(analogPin)` would still work.

But it doesn't!
Setting the `pinMode()` is OK, but once you use `digitalRead()' once ***after*** calling `analogRead()` kills any subsequent calls to `analogRead()`.

```cpp
// tested on Photon with v0.5.0
#define aPIN A0
#define INTERFERE_ONCE  true
#define INTERFERE_MULTI false

void setup()
{
  pinMode(D7, OUTPUT);
  pinMode(aPIN, INPUT_PULLUP);  // set this once as default state for the pin
  if (INTERFERE_ONCE)
  {
    analogRead(aPIN);
    delay(100);
    digitalRead(aPIN);
  }
}

void loop()
{
  int b;
  if(INTERFERE_MULTI)
  {
    b = digitalRead(aPIN);
    digitalWrite(D7, b);
    Serial.println(b);
    delay(100);
  }

  // aPIN should still be set as INPUT_PULLUP
  Serial.println(analogRead(aPIN));  // will be temporarily changed to AN_INPUT
  // and now should be back to INPUT_PULLUP for next time round

  delay(1000);
}
```
---

Completeness:
- [x] Minimum test case added
- [x] Device, system and user firmware versions stated
- [ ] Particle confirmed

ScruffR, 2016-05-18:
No confirmation of this issue yet?

avtolstoy, 2016-05-22:
Issue confirmed and fixed in #1006. DAC pins may also be affected by the issue under certain conditions.

PR #1006 (2016-06-10): digitalRead()/analogRead() interference
Fixes #993

- Saves PinMode in `PIN_MAP[pin].user_property` instead of a single global variable
- Fixes ADC PinMode configuration
- Fixes DAC PinMode configuration (affected by the same issue as ADC pins described in #993 under certain conditions)

Relevant tests:
- `wiring/no_fixture` `GPIO_AnalogReadWorksMixedWithDigitalRead`
- `wiring/adc_dac` `DAC_AnalogWriteWorksMixedWithDigitalRead`

---

Doneness:

- [x] Contributor has signed CLA
- [x] Problem and Solution clearly stated
- [ ] Code peer reviewed
- [ ] API tests compiled
- [ ] Run unit/integration [... cut]
closed 2016-06-10 by m-mcgowan