particle-iot/device-os #1007: Define Platform IDs somewhere

embedded-creations, 2016-05-24:
It would be nice if there were definitions made for the various PLATFORM_IDs, as right now you have to add a comment if you want platform specific code to be readable.  e.g. 

`#if PLATFORM_ID == 0 // Core`

In many cases the comments get left out, and you're left with this, which isn't readable without documentation or a good memory:

`#if PLATFORM_ID == 0`

This would be much better:

`#if PLATFORM_ID == PLATFORM_ID_CORE`

m-mcgowan, 2016-05-24:
There is! :-) It's here

https://github.com/spark/firmware/blob/develop/platform/shared/inc/platforms.h

embedded-creations, 2016-05-24:
Sweet!  I guess not many people know about it.  There's no search results on the community for "PLATFORM_SPARK_CORE", but plenty of results for "PLATFORM_ID" all using magic numbers.

platforms.h didn't come up when I did a search of the firmware folder for "PLATFORM_ID", could you add a comment to it?  e.g. `// Use these definitions with PLATFORM_ID`

closed 2016-05-24 by embedded-creations