particle-iot/device-os #1008: sleep(pin, edge, timeout) interferes with Safe and DFU Mode

ScruffR, 2016-05-25:
I've had this funny behaviour on my Photon with 0.5.1-rc.2 that having my device slept once with this code
```cpp
const char evtName[] = "scruffy_test";

//--SYSTEM_MODE(MANUAL);
#define WATER_SENSOR D0

// Spark setup
void setup()
{
  pins_init(); 
}

void loop()
{
  int water = isExposedToWater();
  Particle.publish(evtName, water ? "on" : "off", PRIVATE);
  delay(5000);
  System.sleep(WATER_SENSOR, FALLING, 30);
}

// initialize our pins
void pins_init()
{
  pinMode(WATER_SENSOR, INPUT_PULLUP);
}

// determine if we're exposed to water or not
boolean isExposedToWater()
{
  return !digitalRead(WATER_SENSOR);
}
```

I can't just enter Safe or DFU mode anymore. 
The RGB LED does the usual stuff till I release SETUP but then just resets and starts my code.

However, if I hold SETUP through yellow the device resets goes back to magenta (respectively yellow if held longer) and when I let go on the second round the device enters the desired mode.

---

Completeness:
- [X] Minimum test case added
- [X] Device, system and user firmware versions stated
- [ ] Particle confirmed

ScruffR, 2016-07-29:
No Particle confirmation?

m-mcgowan, 2016-09-13:
Are you hitting the reset button at anytime? This is needed since the CPU is still actually executing your code - it's just paused in the sleep function.

ScruffR, 2016-09-13:
It's so long, so I can't remember, but could check again.
But considering that the device wouldn't wake on pressing SETUP alone to play that game, I'd suspect I did press RESET too.
And when I said: "... having my device slept once ...", I'd even think that I wouldn't expect the device to enter Safe or DFU mode without tapoing RESET while running.

technobly, 2017-02-07:
I tested with 0.5.1 and could not reproduce.  All functions worked as intended with the provided code.  Please reopen if this is still an issue :)

closed 2017-02-07 by technobly