particle-iot/device-os #1012: 2G Electron not updating user code OTA or via serial - not reporting any errors

bblack7489, 2016-05-30:
Please see this thread on the community:
https://community.particle.io/t/electron-2g-breathing-cyan-but-apparently-not-executing-code/23071

As a summary, I have a 2G Electron and Windows Surface Book running Win 10.  I have updated the firmware on the electron, I have been testing with a fully charged battery and have good cellular coverage.

I can only update the user code on my Electron via the CLI and specifically only via the --usb interface.  Using --serial or trying to update OTA returns a successful response but does not change the software running on the device.  It will update, reboot and then continue to run the previous version of the application.

---

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

monkbroc, 2016-06-02:
This is really bizarre. Let me know if this resolves itself in the future. I'm developing on Windows 10 and I don't have this issue with serial flashing an Electron 2G.

bblack7489, 2016-06-02:
Yes, it's very strange.

I submitted a support ticket as well, and last night they gave me a couple
of things that I could try.  I haven't had a chance to do them though, but
maybe that will give me more information.

Is there a hardware reason on the Electron itself that would cause serial
and OTA updating to fail?  Without knowing the details of the hardware or
firmware layers a little better, I'm just grasping at straws as to what the
two methods might have in common.

~Ben

technobly, 2016-06-02:
@bblack7489 now that you have the DFU-UTIL working and Particle Update completed successfully, I'm guessing you don't have any issue with your system firmware.  That said, what version of the CLI are you running? `particle --version`  Also now try updating via the Build IDE and see if the OTA flash takes effect.  DFU transfer over USB uses the bootloader to write the flash memory directly, version OTA which first copies the flash binary to a separate section of flash memory and resets the device.  When it boots again it knows there is a new image waiting for it and the bootloader copies it to the correct spot.  It's possible if your OTA section of memory is damaged somehow, like maybe via ESD event, then OTA binaries are not quite valid.  I'd have to double check, but I believe the SERIAL transfer uses the same OTA flash memory and reboot process, which may explain why both of those methods are different than DFU over USB.  If you have access to JTAG tools, we can dig in a little bit and see if various sectors of flash are unprotected as they should be.  Then we could attempt to erase and restore your device.  That should tell us if flash is corrupted.  If you don't have these tools, you should request a replacement device from support.  Please close this issue if you find a successful path that seems to indicate a hardware failure.  Thanks!!

bblack7489, 2016-06-03:
Alright, I'm using CLI version 1.11.0.

I'm sorry, I don't have JTAG tools at my disposal, so maybe it's time to
request a replacement.

I'm glad to see that this is likely a hardware problem and not a firmware
issue.

~Ben

technobly, 2016-06-03:
@bblack7489 you need at least v1.12.0 to get v0.5.0 system firmware for the Electron with `particle update` and v1.14.2 to get v0.5.1 system firmware.

technobly, 2016-06-03:
@bblack7489 more info, please update your CLI again with `npm update -g particle-cli` and check the version, it should be 1.14.2 as of right now.  Then try the `particle update` again and see if things change for you.

bblack7489, 2016-06-03:
Brett I was definitely hoping that this worked.  I updated to 1.14.2 and
updated the particle firmware to 0.5.1, but there is still no change in the
OTA or serial flashing.

Any other thoughts before I start the return ticket?

~Ben

technobly, 2016-06-03:
yeah it does sound like a hardware issue.  Please mention to Customer Support that I would like to do a post mortem on your Electron.  Thanks!

m-mcgowan, 2016-06-23:
If the user app doesn't update via Serial or OTA, it most likely doesn't have a valid checksum. (Dfu doesn't validate the app.)

Please run `particle binary inspect firmware.bin` and double check that the binary file is valid.

jakeypoo, 2016-06-27:
I wanted to chime in, I have two P1 units that are experiencing similar issues with OTA updates. 
Details:  
- Unable to flash user-part or system-part binaries 
- Unit 1 is running 0.5.0 firmware
- Unit 2 is running 0.4.9 firmware
- Both are running stable contacting my server and contacting the particle cloud
- Other identical devices in the same location are able to flash the same binary OTA just fine

What happens : 
1) I flash the file either via the particle CLI or via the web-api 
2) I get an 'update started' message from the CLI / 200 ('update started')  response from the api call
3) The device goes offline for a bit as if it were restarting after a flash.
4) When it gets back online it is still running the previous firmware (not updated, doesn't revert to tinker)

Unfortunately the devices are remote and I have no way of getting more details for now.

closed 2020-02-29 by avtolstoy