particle-iot/device-os #1015: Reset IP mode to automatic when clearing credentials

monkbroc, 2016-06-01:
I had devices that were not connecting to the Particle cloud because they had a static IP set by a previous program. I reset WiFi credentials but that did not reset the IP mode.

`wlan_clear_credentials` should also set the `static_ip_config_t.config_mode` back to DHCP.

---

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

m-mcgowan, 2016-06-04:
I'm not sure about this, since it seems to be violating the "element of least surprise". E.g. we don't reset the antenna selection when clearing wifi credentials. (If we reset the IP setup then why not also reset other things related to wifi?)

I would instead rather have a "super Wi-Fi reset" command that resets all persisted WiFi state:
- antenna
- IP configuration
- credentials
- country code

The super wifi reset could be triggered by a serial command, or SoftAP command. 

How does that sound?

monkbroc, 2016-06-04:
I was surprised for 2 days why my devices were not connecting to my home WiFi anymore.

I still think that IP config is strongly tied to WiFi credentials. For example on Linux the IP config (DHCP/manual IP) and WiFi settings are in the same dialog box called "Edit connection information".

The antenna is a hardware setting that won't need to change unless you modify your board.
The country code should be set and forget for most people.
On the other hand if I go from a place where I need a fixed IP back to my home with DHCP, I won't be able to connect by clearing WiFi credentials and entering my home WiFi credentials.

technobly, 2016-06-05:
I tend to agree that it should be reset to DHCP as well with credentials, because this puts it back into an AUTOMATIC mode.  Because there is no easy way to discern if the device is configured for a static IP, when things go wrong your tendency will be to try to reset things.  As long as we agree that DHCP should always result in an automatic connection process, then resetting this way when you were previously connected to a STATIC IP should still connect.  Let's explore a bit further though along this logic.

The antenna setting is one of those things that if you designed your product in a metal box with an external antenna, resetting to an AUTO state would still be safe... That's not quite how we default the antenna anymore due to issues with how WICED interprets AUTO so we default to INTERNAL.  In this case a reset would be fatal if the device is remote, so we don't make that easy to reset.

Painting it in that light, could resetting STATIC IP back to DHCP for a remote device be fatal?

andyw-lala, 2016-06-06:
Yes, very much so.