particle-iot/device-os #1016: Particle.subscribe() with "MY_DEVICES" parameter issue

XuGuohui, 2016-06-06:
Hi there,

I have two RedBear Duo devices both claimed on Particle Build. I used a Duo to publish an button pressed event and another one to subscribe this event. 

1. When I publish the event as public, then the `Particle.subscribe( event, handler )` can successfully subscribe this event, while the `Particle.subscribe( event, handler, MY_DEVICES )` did not.

2. When I publish the event as private, then the `Particle.subscribe( event, handler, MY_DEVICES )` can successfully subscribe this event, while the `Particle.subscribe( event, handler)` did not.

I was thinking I can subscribe my devices events no matter it is published as public or private, and no matter if I subscribe it with the `MY_DEVICES` parameter or not, because it is my device.

Now the `MY_DEVICES` seams like acting as the `MY_PRIVATE` role -- It block the public events published by my device.

My wrong understanding?

Although I was using the Duo to test my code, but I think the cloud function should be compatible with Photon.

A Duo publishs event:
```
int button = D0;

void setup() {
  Serial.begin(115200);
  Serial.println("Application started!");

  pinMode(button, INPUT_PULLUP);
}

void loop() {
  // Publish an event with data, with user specified TTL.
  if(digitalRead(button) == LOW) {
    delay(50); // Debounce
    if(digitalRead(button) == LOW) {
      if (!Particle.publish("button-pressed", "LOW", 21600)) {
        Serial.println("Publish 'button-pressed' event failed.");
      }
      Serial.println("Publish 'button-pressed' event successfully.");
      while(digitalRead(button) == LOW); // Wait to release.
    }
  }
}
```

Another Duo subscribes this event:
```
int btn_event_cnt = 0;

void myBtnEvtHandler(const char *event, const char *data);

void setup() {
  Serial.begin(115200);
  Serial.println("Application started.");

  if(!Particle.subscribe("button-pressed", myBtnEvtHandler, MY_DEVICES)) {
    Serial.println("Subscribe to event 'button-pressed' failed.");
  }
  Serial.println("Subscribe to event 'button-pressed' successfully.");
}

void loop() {
  // Do nothing.
}

void myBtnEvtHandler(const char *event, const char *data) {
  btn_event_cnt++;
  Serial.print(btn_event_cnt);
  Serial.print(" ");
  Serial.print(event);
  Serial.print(", data: ");
  if (data)
    Serial.println(data);
  else
    Serial.println("NULL");
}
```

Any suggestion would be appreciated!

---

Completeness:
- [x ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

ScruffR, 2016-07-29:
That's the intended behaviour by design

XuGuohui, 2016-07-29:
Thanks for the clarification!

technobly, 2017-02-07:
Added a Docs issue here https://github.com/spark/docs/issues/560

closed 2017-02-07 by technobly