particle-iot/device-os #1017: USART serial availableForWrite() not working

rickkas7, 2016-06-06:
There’s a simple typo in the implementation of `USARTSerial::availableForWrite()`:

https://github.com/spark/firmware/blob/develop/wiring/src/spark_wiring_usartserial.cpp#L69

It should probably be `HAL_USART_Available_Data_For_Write`. However, if I fix that and use it with code like:

```cpp
        Serial.printlnf("%d", Serial1.availableForWrite());
        Serial1.println("01234567890123456789012345678901234567890123456789");
        Serial.printlnf("%d", Serial1.availableForWrite());
```

The output is:

```
0
50
```

I’m unclear of how the function is supposed to work. Should it:

1. Return the number of bytes waiting to be written out from the buffer (0=empty send buffer), which is what it does.
2. Return the number of bytes you can write to the buffer without blocking, which would be SERIAL_BUFFER_SIZE - the value currently returned.

Thanks,
Rick

---

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

m-mcgowan, 2016-06-07:
@avtolstoy Any thoughts on this?

avtolstoy, 2016-06-07:
Confirmed and fixed in #1020 :
1. A typo in `USARTSerial::availableForWrite()`
2. `availableForWrite()` behavior, which should behave as described in docs:
> Retrieves the number of bytes (characters) that can be written to this serial port without blocking.

rickkas7, 2016-06-07:
Thanks! Sorry, I swear I looked it up in the docs and didn’t see it, but yes, the correct behavior is documented.

PR #1020 (2016-06-10): Fixes USARTSerial::availableForWrite() behavior
Fixes #1017 

`wiring/serial_loopback` includes a test `SERIAL1_AvailableForWriteWorksCorrectly` to verify correct `availableForWrite()` behavior.

---

Doneness:

- [x] Contributor has signed CLA
- [x] Problem and Solution clearly stated
- [x] Code peer reviewed
- [x] API tests compiled
- [x] Run unit/integration/application tests on device
- [x] Add documentation
- [ ] Add to CHANGELOG.md after merging (add links to docs and issues)
closed 2016-06-07 by rickkas7