particle-iot/device-os #1018: Queue feature, compile error - Dynimical Library issue.

Darma, 2016-06-07:
Hi,

I am working around to use Queue-ing feature of the Photon Firmware, from both [latest](https://github.com/spark/firmware/tree/latest) and [development](https://github.com/spark/firmware/tree/89029fb0faf4b7b6cfe91543181d457a0e24c572) branch.

For example:
```Cpp
os_queue_t my_queue; 
os_queue_create( &my_queue, 10, 10);
```
when compile I get following error message:

```
.. undefined reference to `os_queue_create'`
```
It seems like the source code is already in the relevant files: 

> /firmware/hal/inc/concurrent_hal.h 
> /firmware/hal/src/stm32f2xx/concurrent_hal.cpp

But it looks like it's not implemented (linked) in the dynamical library  `hal_dynalib_concurrent.h` file. At least current [development branch](https://github.com/spark/firmware/blob/89029fb0faf4b7b6cfe91543181d457a0e24c572/hal/inc/hal_dynalib_concurrent.h).  
> /firmware/hal/inc/hal_dynalib_concurrent.h

**Question:**
- How can or do I use Queue (os_queue*)  functions?

---

Completeness:
- [x] Minimum test case added
- [x] Device, system and user firmware versions stated
- [ ] Particle confirmed

m-mcgowan, 2016-06-07:
Thanks for bringing this up! At present the queues are intended only for use by the system, since it's quite a low-level interface. I can add that low-level interface to the dynalib for the 0.6.0 release, and then we'll add a high-level wiring `Queue` interface for 0.7.0

m-mcgowan, 2016-06-07:
Please see the PR above - you can pull that into your local working copy. Then recompile system-part2 and flash that to your device, then recompile the app and the low-level queue interface will be available.

closed 2016-06-10 by m-mcgowan

Darma, 2016-06-20:
Hey thanks, when will 0.6.x released? Any time estimation?

aeris-ming, 2018-07-16:
Please update document If you guys do not want to export Free RT-OS API.
My suggestion is follow GUN style not stupid Aduino.

PR #1019 (2016-06-10): low-level queue API
makes the os_queue interface extensible, and exposes it to the dynamic library.

fixes #1018 

---

Doneness:

- [x] Contributor has signed CLA
- [x] Problem and Solution clearly stated
- [x] Code peer reviewed
- [x] Add to CHANGELOG.md after merging (add links to docs and issues)