particle-iot/device-os #1023: Particle.publish() causes device to disconnect regularly

ScruffR, 2016-06-09:
https://community.particle.io/t/connecting-particle-to-google-doc/23543/18
In this thread I couldn's see an issue with the code till I actually tested and realised that there must be something wrong with the rate limit of `Particle.publish()`.
Although we are allowed one publish per second, going close - but not over - that limit causes frequent cloud disconnect/reconnect glitches.

With this code on a Photon running 0.5.1 I get consistent reconnects every 60sec.
```cpp
const uint32_t msPublish = 1000;

void setup() {
}

void loop() {
    static uint32_t ms;

    if (millis() - ms > msPublish)
    {
        ms = millis();
        Particle.publish("ScruffyTest", "Dummy!", PRIVATE);
    }
}
```

But when I increase `msPublish = 1100` the problem goes away.

---

Completeness:
- [X] Minimum test case added
- [X] Device, system and user firmware versions stated
- [ ] Particle confirmed

ScruffR, 2016-07-29:
No Particle confirmation?

technobly, 2016-07-30:
Hey @scruffr I can confirm this behavior. We have already made some cloud side fixes that should have helped, and more device side fixes to come. We are tracking internally and I'll add the issue to github.

technobly, 2017-02-07:
A value of 1000 is confirmed to no longer cause disconnects.

closed 2017-02-07 by technobly