particle-iot/device-os #1025: [Electron] Flashing blue but RESET doesn't cure it

ScruffR, 2016-06-10:
This is not a duplicate but an "escalation" of this issue, since that issue was RESET resolvable, but mine isn't
 https://github.com/spark/firmware/issues/687

The device got stuck in SETUP mode, and a RESET push just put it into short breathing white and then SETUP again. Only unplugging the LiPo helped, but nothing else was required (no SIM wiggling, no antenna action, just repower).

I'm running this on an Electron G2 v0.5.1 (sorry can't really cut it down to test, since the error is very sporadic, so finding a failing minimal sketch would take ages ;-) )
```cpp
SYSTEM_MODE(MANUAL)
STARTUP(System.enableFeature(FEATURE_RETAINED_MEMORY))
// Dashoard: https://app.ubidots.com/ubi/public/getdashboard/page/Ked8IDJliKBciQci4MuRh865ACA

#include "DS18B20/Particle-OneWire.h"
#include "DS18B20/DS18B20.h"
#include "PietteTech_DHT/PietteTech_DHT.h"
#include "Ubidots/Ubidots.h"
#include "Adafruit_SSD1306/Adafruit_GFX.h"
#include "Adafruit_SSD1306/Adafruit_SSD1306.h"

// If using software SPI (the default case):
#define OLED_GND    D2
#define OLED_VCC    D3
#define OLED_CLK    D4
#define OLED_MOSI   D5
#define OLED_RESET  D6
#define OLED_DC     D7
#define OLED_CS     -1  // active LOW via GND

#define TOKEN "Put here your Ubidots TOKEN"
#define DATA_SOURCE_NAME "Particle" 
#define DATA_SOURCE_TAG  "Temps"  

#define DHTTYPE  DHT22       // Sensor type DHT11/21/22/AM2301/AM2302
#define DHTPIN   A1          // Digital pin for communications

const uint32_t PUBLISH_MINUTES  = 15; 
const uint32_t SAMPLE_INTERVAL  = 10000;
const uint32_t PUBLISH_INTERVAL = PUBLISH_MINUTES + 60000;
const int led = D7;

void dht_wrapper(); // must be declared before the lib initialization

PietteTech_DHT DHT(DHTPIN, DHTTYPE, dht_wrapper);
DS18B20 ds18b20(A0); 
Ubidots ubidots(TOKEN);
FuelGauge fuel;
//Adafruit_SSD1306 display(OLED_MOSI, OLED_CLK, OLED_DC, OLED_RESET, OLED_CS);

char szInfo[64];
retained double   tempWater    =  0;
retained double   tempAir      =  0;
retained double   relHumidity  =  0;
retained double   dewPoint     =  0;
retained double   dewPointSlow =  0;
retained double   SoC          =  0;
retained int      lastHour     = 24;

void setup() 
{
    #if (PLATFORM_ID == PLATFORM_ELECTRON_PRODUCTION)
      Cellular.on();
      Cellular.connect();
    #else
      WiFi.on();
      WiFi.connect();
    #endif

    if (Time.hour() < lastHour)
    {
        Particle.connect();
        Particle.syncTime();
    }
    delay(1000);
    ubidots.setDatasourceName(DATA_SOURCE_NAME);
    ubidots.setDatasourceTag(DATA_SOURCE_TAG);
}

void loop() 
{
    getTemp();
    publishData();
}

void publishData()
{
  if (!ds18b20.crcCheck()) return;
  
  ubidots.add("WaterTemp"  , tempWater  );
  ubidots.add("AirTemp"    , tempAir    );
  ubidots.add("RelHumidity", relHumidity);
  ubidots.add("DewPoint"   , dewPoint   );
  ubidots.add("SoC"        , SoC        );
  ubidots.sendAll();

  sprintf(szInfo, "Water %.1f °C, Air %.1f °C, Humidity %.1f% %, DewPoint %.1f (%.1f) °C, SoC: %.1f% %", tempWater, tempAir, relHumidity, dewPoint, dewPointSlow, SoC);
  Serial.println(szInfo);
//  Particle.publish("scruffyTempInfo", szInfo, PRIVATE);

  for(uint32_t ms = millis(); millis() - ms < 10000;)
    Particle.process();

  int dt = (PUBLISH_MINUTES - Time.minute() % PUBLISH_MINUTES) * 60 - Time.second();  // wake at next time boundary
  lastHour = Time.hour();
  System.sleep(SLEEP_MODE_DEEP, dt, SLEEP_NETWORK_STANDBY);   
  //System.sleep(D2, FALLING, dt, SLEEP_NETWORK_STANDBY);   
}

void getTemp()
{
  if(!ds18b20.search())
  {
    int dsAttempts = 0;
    double _tempWater;
    ds18b20.resetsearch();
    _tempWater = ds18b20.getTemperature();
    while (!ds18b20.crcCheck() && dsAttempts < 4)
    {
      Serial.printf("Bad value (%d)", dsAttempts++);
      if (dsAttempts == 3)
        delay(1000);
      ds18b20.resetsearch();
      _tempWater = ds18b20.getTemperature();
    }
    Serial.println(_tempWater);

    if(dsAttempts < 4)
    { // we have a valid reading
      tempWater = (_tempWater + tempWater) / (tempWater ? 2.0 : 1.0);
    }
  }

  switch (DHT.acquireAndWait()) 
  {
    case DHTLIB_OK:
      Serial.println("DHT OK");
      relHumidity  = (DHT.getHumidity()     + relHumidity ) / (relHumidity  ? 2.0 : 1.0);
      tempAir      = (DHT.getCelsius()      + tempAir     ) / (tempAir      ? 2.0 : 1.0);
      dewPoint     = (DHT.getDewPoint()     + dewPoint    ) / (dewPoint     ? 2.0 : 1.0);
      dewPointSlow = (DHT.getDewPointSlow() + dewPointSlow) / (dewPointSlow ? 2.0 : 1.0);
      break;
    default:
      Serial.println("Some error - who cares?");
      break;
  }
    
  SoC = fuel.getSoC();
}

void dht_wrapper() 
{
  DHT.isrCallback();
}
``` 

---

Completeness:
- [X] Minimum test case added
- [X] Device, system and user firmware versions stated
- [ ] Particle confirmed

ScruffR, 2016-06-16:
This bug starts to get annoying
As this screenshot shows
[image]

I had three of these incidents, but the main annoyance about it is that that Listening Mode keeps the device awake, draining my battery much quicker than expected and doesn't even give the device a chance to recover (a replaced a drained battery for a fulky charged one just to find it completely drained again the next morning :rage:)
I'm in a spot where the German and Austrian network often fight and I keep getting networks switches on my phone regularly, so the reason why a connection might fail is not always the device's fault, but keeping it in this non-recoverable state will not help anybody.

vidiot1969, 2016-07-06:
Same device and firmware here, and also using PietteTech_DHT/PietteTech_DHT.h, which I mention just in case.  It is sporadic for me, and has affected me in much the same way -- i.e. it doesn't always happen, Reset usually helps, but not always.  I've been purposely charging and draining my battery to observe the behaviour, since this is when I have noticed this happen most.

I am using the JS SDK, and error 400 is reported in the browser console.

vidiot1969, 2016-07-29:
@ScruffR , did you make any progress with this issue?  I continue to experience problems once battery drain occurs, and carefully observed the "short breathing white and then SETUP" modes you mention all day yesterday.  RESET did not help.  2G here as well.

ScruffR, 2016-07-29:
I have decided to use `System.sleep(WKP, RISING, timeout, SLEEP_NETWORK_STANDBY)` insread and haven't had problems since.
But the original issue still exists and seems unseen (not confirmed) by Particle.

ScruffR, 2016-08-30:
Will this ever be addressed? Maybe in connection with https://github.com/spark/firmware/issues/687

technobly, 2017-02-07:
This has been fixed in 0.6.1-rc.1 by providing a default timeout of 5 minutes for Listening Mode, and also making it user configurable.  https://prerelease-docs.particle.io/reference/firmware/electron/#setlistentimeout-

technobly, 2017-02-07:
This has been fixed in 0.6.1-rc.1 https://github.com/spark/firmware/pull/1182 by providing a default timeout of 5 minutes for Listening Mode, and also making it user configurable.  https://prerelease-docs.particle.io/reference/firmware/electron/#setlistentimeout-

closed 2017-02-07 by technobly