particle-iot/device-os #1028: [Electron] waitFor() does not work as expected

ScruffR, 2016-06-12:
As discussed in this thread
https://community.particle.io/t/system-mode-system-threading-and-loop-waitfor-waituntil/23661/3

This code does not continue execution after the set timeout, but waits full five minutes before carrying on (Electron U270, 0.5.1 - with detached antenna)
```cpp
// How long to wait for a network connection
#define WAIT_FOR_CONNECTION_SECONDS 60

// Only connect when it is required
SYSTEM_MODE(SEMI_AUTOMATIC);

// Connect to the network
static void connect() {
    time_t timeStarted = Time.now();
   
    if (!Cellular.ready())
    {
      Serial.printf("Connecting to cell tower at %s (Time.now() %d seconds), waiting for up to %d seconds...\r\n", Time.timeStr().c_str(), Time.now(), WAIT_FOR_CONNECTION_SECONDS);
      Cellular.connect();
      if (waitFor(Cellular.ready, WAIT_FOR_CONNECTION_SECONDS)) {
        Serial.printf("Connected at %s (Time.now() %d seconds, connection process took %d seconds).\r\n", Time.timeStr().c_str(), Time.now(), Time.now() - timeStarted);
      } else {
        Serial.printf("WARNING: connection failed at %s (Time.now() %d seconds, timeout was %d seconds).\r\n", Time.timeStr().c_str(), Time.now(), Time.now() - timeStarted);
      }
    } else {
      Serial.println("Already connected to cell tower.");
    }

    timeStarted = Time.now();
    if (!Particle.connected()) {
        Serial.printf("Connecting to network at %s (Time.now() %d seconds), waiting for up to %d seconds...\r\n", Time.timeStr().c_str(), Time.now(), WAIT_FOR_CONNECTION_SECONDS);
        Particle.connect();
        if (waitFor(Particle.connected, WAIT_FOR_CONNECTION_SECONDS)) {
            Serial.printf("Connected at %s (Time.now() %d seconds, connection process took %d seconds).\r\n", Time.timeStr().c_str(), Time.now(), Time.now() - timeStarted);
        } else {
            Serial.printf("WARNING: connection failed at %s (Time.now() %d seconds, timeout was %d seconds).\r\n", Time.timeStr().c_str(), Time.now(), Time.now() - timeStarted);
        }
    } else {
        Serial.println("Already connected to network.");
    }
}

void setup() {
    // Opens up a Serial port so you can listen over USB
    Serial.begin(115200);
    while(Serial.read() < 0) Particle.process();
    pinMode(D7, OUTPUT);
    digitalWrite(D7, HIGH);
    Serial.println("Let's go!");
    
    Cellular.on();
    
    // Set time to zero when we start so that we know if time
    // is synchronised or not later on
    Time.setTime(0);
    
    // Connect to the network so as to establish time
    connect();
}
```

---

Completeness:
- [X] Minimum test case added
- [X] Device, system and user firmware versions stated
- [ ] Particle confirmed

m-mcgowan, 2016-06-12:
waitFor() timeouts are accurate only in multithreaded mode, otherwise they are at the mercy of the single threaded execution. On the electron, connections to the tower can take 5 minutes to complete.

ScruffR, 2016-06-12:
OK, then this fact might need a note in the docs to clarify this.
But the OP stated this (not tested myself yet tho')

> If I enable threading then waitFor() doesn't do any waiting at all and so I get:
> Connecting to network at Thu Jan  1 00:00:05 1970 (Time.now() 5 seconds), waiting for up to 60 seconds...
> WARNING: connection failed at Thu Jan  1 00:00:05 1970 (Time.now() 5 seconds, timeout was 0 seconds).

So even with `SYSTEM_THREAD(ENABLED)` it doesn't seem to work accurately.

GeertWille, 2016-07-27:
I also confirm having issues with waitFor timeout not kicking in... I do have `SYSTEM_THREAD(ENABLED)` but the program is not continuing after the timeout.

technobly, 2016-09-13:
@ScruffR can you try it with with 60000 timeout instead of 60.  That might be why it appears not to wait long enough.

ScruffR, 2016-09-13:
Doh', I'll have to check that.

I just got so distracted by the unexpected 5min timeout without `SYSTEM_THREAD(ENABLED)` (which is still a bit confusing IMHO), that I completely missed that ms vs. sec.

technobly, 2017-02-07:
@ScruffR did you ever get a chance to check the 60 vs 60000 issue?  I've noted that we need to improve documentation as well.

ScruffR, 2017-02-07:
I haven't had issues lately, but I must admit I have heeded Matthews advice and always used `SYSTEM_THREAD(ENABLED)` ever since.
But I'll try.

technobly, 2017-02-07:
I only ask because you said "So even with SYSTEM_THREAD(ENABLED) it doesn't seem to work accurately." and you were using waitFor to wait for cellular to connect, the handshake to complete and be connected to the Cloud.  So multithreaded or not, you are going to have to wait at least 30-120 seconds.  I would `#define WAIT_FOR_CONNECTION_SECONDS 300000` to ensure you wait at least as long as the system is trying to connect.

ScruffR, 2017-04-26:
Reviving an old issue ;-)
This might be connected or not.

In this thread we saw a wrong positive on `Cellular.ready()` (in a `waitFor(Cellular.ready)` call) that was due to incorrect readiness indication after `Cellular.off() -> Cellular.on() -> Cellular.connect()`.

closed 2020-02-29 by avtolstoy