particle-iot/device-os #1034: receipt acknowledgement for events sent with Particle.publish()

m-mcgowan, 2016-06-15:
With the Electron, CoAP responses are not waited for, but are processed in the background. This can mean that Particle.publish() provides only a guarantee that an event was sent, but not necessarily that it was received. 

Some possibilities for the API

- a new flag to `Particle.publish()` that indicates the call should block until the event has been received by the cloud (or a timeout occurs.)

- return from `Particle.publish()` a unique ID for the event, and allow the application to register a listener for all satisfied events

- return a future that indicates if the event has been received or not. (This could be built upon the event listener above.)

```cpp

   bool success = Particle.publish("topic", "data", WAIT_ACK);

   // or

  Particle.publish("topic", "data", CRITICAL);
  // attempts to send the event, and keeps retrying (more than the 4 retries for regular CoAP messages.)

   // or

  System.on(event_delivered,  delivered);
  void delivered(int eventID) {
     // can handle the delivered event
  }

  int eventID = Particle.publish("topic", "data");
  
  // or

  auto event = Particle.publish("topic", "data");
  waitFor(event.delivered, 30000);  
```

This can be combined in a single line, giving the simplicity of a flag with control over how long to wait. 

```
   bool delivered = Particle.publish("topic", "data").waitForDelivery(30000);
```

while it also provides a future to allow for more advanced cases where the application doesn't want to block waiting for the event, but wants to check on the publish status. 

---

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions sstated
- [ ] Particle confirmed

sergeuz, 2016-10-05:
I would go with the approach involving futures, as it allows to support all possible handling options in single unified interface, depending on what is more convenient for an application developer in particular case. For example:

* Synchronous waiting:
```cpp
Particle.publish("topic", "data").wait();
```
* Registering completion callback:
```cpp
auto f = Particle.publish("topic", "data");
f.onSuccess(okCallback); // okCallback()
f.onError(errorCallback); // errorCallback(const Error&)
```
* Error handling (synchronous):
```cpp
auto f = Particle.publish("topic", "data");
f.wait();
if (f.hasError()) {
    Log.error("Unable to publish event: %s", f.error().message());
}
```

m-mcgowan, 2016-10-05:
this looks good 👍  

It's not clear to me if we should change the default publish to be async or fully synchronous  - currently it's fully synchronous on the photon, but only half-synchronous on the electron since it waits for the initial message to be sent, but doesn't wait for guaranteed delivery.  @jenesaisdiq - any thoughts on if we should change the default behaviour?

closed 2016-11-29 by technobly