particle-iot/device-os #1036: gcc virtual device runs at 100% cpu

m-mcgowan, 2016-06-15:
The gcc virtual device emulates an embedded device, where the CPU is being given work to do all the time. This carries over to the cross compiled version, where it consumes 100% of the executing cpu core on a desktop machine.

Some steps to mitigate this will make the cross compile device more scalable and environmentally friendly.

---

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

sergeuz, 2016-06-15:
As a simple solution we can try to call `std::this_thread::yield()` before reentering busy loops, such as the application loop. Probably this can be even automated via gcc's instrumentation.

joegoggins, 2016-06-15:
A non firmware way to do this is to run a gcc virtual device in a Docker container and specify the `--cpu-shares` (relative CPU shares)`, `--cpu-period`, or `--cpuset-cpus` flags to the [docker run command](https://docs.docker.com/engine/reference/run/) to limit how much CPU it consumes.

andyw-lala, 2016-06-15:
That's just going to slow down the instance running in the container. It
will solve the host resource problem of pegging a core, but it will make
the instance run strangely as it meters the cpu out to the container.

closed 2016-11-24 by m-mcgowan

hazeem1991, 2019-06-28:
i really having like this kind of problem but i did not find any way to solve it any advice ?

joegoggins, 2019-06-28:
The referenced [PR above provides a solution](https://github.com/particle-iot/device-os/pull/1167) and more advice @hazeem1991

PR #1167 (2016-11-24): Workaround for 100% CPU usage problem on the virtual device
This PR fixes #1036 by suspending main thread for some minimum time on every Nth iteration of the application loop. That's probably not the best solution possible, but it allowed to reduce CPU usage from 100 to 10-15% on my machine (Intel Core i7), while keeping reasonable performance comparing to the real devices.

Note that calling `std::this_thread::yield()` has appeared to be insufficient to workaround the problem, also a simple unconditional sleep on each iteration makes an application significantly slower comparing to the same a [... cut]