particle-iot/device-os #1037: CLI-assisted logging

sergeuz, 2016-06-17:
As we discussed previously, I would be nice to have possibility to enable logging dynamically, via the cloud or over the wire. Taking into account that all currently implemented [log handlers](https://github.com/spark/firmware/blob/develop/docs/reference/firmware.md#log-handlers) already require serial connection with the host, it seems natural to start with the USB logging first, and then reuse the same configuration code for cloud logging.

The plan is to use [vendor-specific USB requests](https://github.com/spark/firmware/pull/1010) (already in `develop`) to configure logging dynamically, without any prior configuration done at application side, and without entering the listening mode, which affects normal workflow of the system and application code greatly. The CLI then can be used to attach to already running system, for example:

```
# Start logging with default settings (uses USBSerial1)
particle log

# Enable verbose logging
particle log all

# Enable verbose logging for application code only
particle log warn app=all

# Use hardware serial and UART to USB converter for logging
particle log -s serial1 -d /dev/ttyUSB0
```

On any of the above commands the CLI sends necessary configuration requests to the device and then starts to read and display a nicely formatted and highlighted logging output.

Implementation-wise, the `LogManager` class needs to be extended to support named and dynamically allocated log handlers. Few factory methods also need to be added in order to allocate log handlers based on their requested type and settings. A trivial JSON formatter for generated logs would also simplify parsing at the CLI side.

Configuration request might look like the following:

```js
{
  "op": "addLogHandler",
  "handler": { # Handler settings
    "type": "JSONLogHandler"
  },
  "stream" : { # Output stream
    "type": "Serial1",
    "baud": 57600
  },
  "level": "warn", # Default logging level
  "filters": {
    "app": "trace" # Category-specific logging level
  }
}
```

I used JSON here for simplicity, though @avtolstoy confirmed that we can use it for vendor-specific requests with no problems, assuming that request size limitations are met.

---

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

m-mcgowan, 2016-06-20:
This is a great start! For the CLI interface, I suggest we follow the same syntax as used for the `flash` command to support both local and remote devices:

```
particle log --usb  ...
```
and
```
particle log my_device ...
```

Using JSON should be fine - we have the `jsmn` json library included in the system firmware. (Although `LogManager` is currently in user space - how shall we bridge the gap from the system-initiated USB vendor requests over to the LogManager in user land? Related to that is how will we be able to use logging when in safe mode, where no application code is presently loaded?)

sergeuz, 2016-06-20:
I was thinking about using `module_user_init_hook()` to install some user space callbacks for my immediate needs, but there are likely better options. I see that `jsmn` functions are exported via part1 dynalib interface, does it mean that its `jsmntok_t` token structure can be considered safe in terms of ABI?

I have not considered the safe mode yet, but assuming that there's no validated user module available, I don't see any option other than instantiating necessary loggers in the part2 (or part3) code. It does look as significant code duplication though :(

m-mcgowan, 2016-06-29:
That's a good question about the jasmin lib, I'd like to say it's ABI stable but I know that newer versions of the jsmn lib have changed structure. Currently, it's only used during listening mode, so we could in principle change the ABI and it would be safe since OTA updates do not enter listening mode between flashing part1 and part2. 

The jsmn api could then be upgraded, and used by logging. We'd also add a jsmn_version() call to allow calling code to check for library compatibility in advance of using it.

closed 2020-02-29 by avtolstoy