particle-iot/device-os #1039: Photon stuck on RED ( SOS) code

bert269, 2016-06-20:
My device was working well, firmware 0.5.0.
 I made a minor change to the code to disable the LED (since the application I want to use it for cannot have the 'breathing LED all the time). I compiled and flashed the code to the Photon. It rebooted and immediately start the SOS sequence of flashes. Very hard to tell the amount of red flashes to determine what is going on.

I cannot seem to get out of this state, even after I reflashed the latest firmware in DFU mode, it flashes white once and immediately goes back to SOS.
I can only get it into DFU mode, nothing else. No other colors, only a brief white, followed by red or yellow color in DFU mode.

Please advise 

Thanks 

---

Completeness:
- [x] Minimum test case added
- [x] Device, system and user firmware versions stated
- [x] Particle confirmed

monkbroc, 2016-06-20:
Hi Bert,

You're going to get better help figuring out the problem in your application and how to revive your Photon if you post to https://community.particle.io. GitHub is best for reporting issues or tracking new features in the Particle system firmware.

m-mcgowan, 2016-09-13:
Sorry to hear of the problems. Did you find a solution for this?

technobly, 2017-02-07:
Also please utilize https://support.particle.io if your issue has not been resolved yet.

closed 2017-02-07 by technobly