particle-iot/device-os #1040: Check that all network calls are done on the system thread

monkbroc, 2016-06-20:
https://github.com/spark/firmware/issues/904 was due to a network function being called on the application thread. Let's audit the system firmware to check that all functions that perform network operations or modify system thread state are called on the system thread.

---

Completeness:
- [ ] Photon network calls reviewed
- [ ] Electron network calls reviewed

m-mcgowan, 2016-09-13:
Any code that is run on the system thread that is not threadsafe, should only ever run on the system thread. This typically means any calls to those functions are wrapped in the SYSTEM_THREAD macro.

We might also consider adding macros to the functions themselves that validate they are run on the correct thread in debug builds, so mistakes can be caught.

m-mcgowan, 2016-09-13:
Some possible approaches:
- get a static analysis of the call stack from the system loop
- get a static analysis of the call stack from all application API entrypoints (use the API testsuite, compiled against gcc.) Can validate all APIs are covered by using code coverage against the `wiring` module.
- if a method on the application thread also exists in the callchain of the system thread then we have a potential issue of calling thread-unsafe code from multiple threads.

PR #1146 (2016-11-29): Asserts for checking that network calls are run on system thread
Closes #1040.
- `Particle.keepAlive()` should be run on system thread
- `DEBUG_THREADING` compilation option added to enable thread asserts
- `ASSERT_ON_SYSTEM_OR_MAIN_THREAD` and `ASSERT_ON_SYSTEM_THREAD` asserts added
- Added `ASSERT_ON_SYSTEM_OR_MAIN_THREAD` and `ASSERT_ON_SYSTEM_THREAD` asserts to `spark_protocol_*` calls

---

Doneness:
- [X] Contributor has signed CLA
- [X] Problem and Solution clearly stated
- [x] Code peer reviewed
- [x] API tests compiled
- [x] Run unit/integration/application tests on device
- [x] Add d [... cut]
closed 2016-11-29 by technobly