particle-iot/device-os #1043: System.sleep() fails with small values of seconds

embedded-creations, 2016-06-22:
With the addition of calls to `network_*` in `system_sleep()`, it now can take more than a few seconds from calling `system_sleep()` to actually entering the sleep mode.  The `HAL_RTC_Set_UnixAlarm()` call is made before the `network_*` calls, so for small values of `seconds`, the RTC alarm will go off before calling `HAL_Core_Enter_Standby_Mode()` and entering the sleep mode.  The end result is that you can call `System.sleep()` with a non-zero value for `seconds` and it will sleep indefinitely.

Here's some demo code (I used 0.5.1 local build to compile) to reproduce this issue, and what I see from my terminal with an Electron running 0.5.1 firmware before it goes to sleep and doesn't wake up.  I'm only using Deep Sleep in my application so I haven't tested behavior with other sleep modes.

https://gist.github.com/embedded-creations/0c22db0f5deddd4f30bc2652ae0e1261

 
```
[Reconnected]
connected, going to sleep for 10 seconds
 
[Closed]
 
[Reconnected]
connected, going to sleep for 9 seconds
 
[Closed]
 
[Reconnected]
connected, going to sleep for 8 seconds
 
[Closed]
 
[Reconnected]
connected, going to sleep for 7 seconds
 
[Closed]
 
[Reconnected]
connected, going to sleep for 6 seconds
 
[Closed]
 
[Reconnected]
connected, going to sleep for 5 seconds
 
[Closed]
```

The minimum `seconds` value needed is variable, I found during one test run of this code that it failed (slept indefinitely) with a sleep value of 10 seconds.

An easy fix might be to make the `HAL_RTC_Set_UnixAlarm()` call right before entering the sleep mode, but that does change the function's behavior, adding some variable amount of time to the total time between calling `System.sleep()` and waking.  A more complicated fix might keep track of the time spent in the function before sleeping.
---

Completeness:
- [x] Minimum test case added
- [x] Device, system and user firmware versions stated
- [ ] Particle confirmed

embedded-creations, 2016-06-29:
It looks like this was updated to be "Particle confirmed" a few days ago, though I would have appreciated even a brief followup message so I could get a notification that someone was looking into this.

I've been doing some more research on the issue as my workarounds of setting a minimum value for `seconds` before going to deep sleep have been failing.  I first tried a minimum value of 30 seconds in my application code, which failed, then increased it to 60 seconds (which also failed).

Looking through the `system_sleep()` code, when using Deep Sleep, there is always going to be a call to `network_suspend()` first.  I can't call `network_suspend()` directly from an application, but I can use `System.sleep(SLEEP_MODE_WLAN, 60)` to call it indirectly without calling the other `network_` functions.

I setup two test Electrons to connect to the cloud, and record the time elapsed to call each `System.sleep(SLEEP_MODE_WLAN, 60)`, `network_disconnect(0, 0, NULL);` and `network_off(0, 0, 0, NULL);` in that order.  One Electron is set to do a `Particle.publish()` directly after connecting, the other doesn't publish.

After letting both run overnight, the time it takes to call `network_disconnect(0, 0, NULL);` and `network_off(0, 0, 0, NULL);` was always 0 seconds.  On the Electron that is not doing a publish, the max time to call `System.sleep(SLEEP_MODE_WLAN, 60)` was 18 seconds.  On the Electron that is doing a publish the max time to call `System.sleep(SLEEP_MODE_WLAN, 60)` was 192 seconds.

Hopefully you can see this is undesirable behavior in two ways: it gives the user less control over battery usage if their Electron may stay awake an extra >3 minutes after telling it to sleep, and it may sleep indefinitely if they tell it to sleep for a "small" number of seconds (currently 192 seconds).

My current workaround in my application is to manually call `network_disconnect(0, 0, NULL);` and `network_off(0, 0, 0, NULL);` before entering Deep Sleep.  I haven't traced through the code but I believe this will skip over any `network_` calls in `system_sleep()` so the processor enters sleep mode quickly.  Please let me know if you have any feedback on this workaround.

Here's the code I used for the test.  You can change the `#define` on line 34 to set if the publish is used or not.  That's the only difference between the two versions of code I used.

https://gist.github.com/embedded-creations/9c8b5c17b5de6e629c319433bd872e7a

avtolstoy, 2016-06-29:
@embedded-creations Thanks for additional info!
I though I wrote a reply after looking through code, sorry for silently confirming this :) There was a similar issue with STOP mode that was fixed in #797, and the same fix wasn't applied to DEEP_SLEEP which is an oversight.

I'm currently looking into this and another System.sleep issue (#1029).

Thanks!

embedded-creations, 2016-06-29:
Great!  Let me know if you need any more tests run.  I seem to be in an area with not the best coverage.  I had an electron [enter listening mode](https://github.com/spark/firmware/issues/687) last night, and another couldn't make a connection in under 90 seconds after many hours of trying last night.

embedded-creations, 2016-06-30:
My workaround doesn't work: even after calling `network_disconnect(0, 0, NULL);` and `network_off(0, 0, 0, NULL);`, the call to `network_process()` can take a long time.  The max I measured was 211 seconds.  I assume that even when the network is disconnected and modem is off, calling `system_sleep()` with an unsent message can cause the modem to turn on and reconnect in an attempt to deliver the message before going to sleep.

My other workaround which I've been testing in parallel is to set a flag in retained memory, reset the Electron, call `Cellular.on()` after seeing the flag is set (so that the firmware knows the modem needs to be turned off before sleeping), then calling `System.sleep(SLEEP_MODE_DEEP, seconds)`.  I'm currently calling `network_disconnect(0, 0, NULL);` and `network_off(0, 0, 0, NULL);` before `System.sleep(SLEEP_MODE_DEEP, seconds)` but I think that's unnecessary.  Anyway, this workaround is complicated, but it's been working across three Electrons, each reporting every three minutes for 24 hours.

I updated the Gist.  The code with Line 56 enabled is what I used to measure the max of 211 seconds.

avtolstoy, 2016-06-30:
@embedded-creations This should be fixed in PR #1051!

embedded-creations, 2016-07-01:
@avtolstoy Great! Looks like a good fix.

I'll either open a new issue or start discussing the "system_sleep() takes as long as 3.5 minutes to sleep" issue in the community so you can close this issue as needed.

PR #1051 (2016-07-20): Sleep fixes
Fixes #1043 and #1029:
- RTC wakeup time should be calculated right before entering SLEEP_MODE_DEEP
- STOP mode should retain user interrupt handler

There are two tests in `wiring/sleep`. Both of them fail on `develop` and pass with this PR.

Tested on Photon and Electron.

---

Doneness:

- [x] Contributor has signed CLA
- [x] Problem and Solution clearly stated
- [x] Code peer reviewed
- [x] API tests compiled
- [x] Run unit/integration/application tests on device
- [ ] Add documentation
- [ ] Add to CHANGELOG.md after merging (add links to docs and issues)

BUGFIX
- System.slee [... cut]
PR #1076 (2016-08-03): [0.5.x] Sleep fixes
Original PR #1051

Fixes #1043 and #1029:
- RTC wakeup time should be calculated right before entering SLEEP_MODE_DEEP
- STOP mode should retain user interrupt handler

---

Doneness:

- [X] Contributor has signed CLA
- [X] Problem and Solution clearly stated
- [x] Code peer reviewed
- [x] API tests compiled
- [x] Run unit/integration/application tests on device
- [x] Add documentation (N/A)
- [ ] Add to CHANGELOG.md after merging (add links to docs and issues)

BUGFIX

- RTC wakeup time now calculated right before entering SLEEP_MODE_DEEP. Fixes [#1043](https://github.com/ [... cut]
closed 2016-07-20 by technobly