particle-iot/device-os #1045: Flashing blue or green following wifi disconnect

pomplesiegel, 2016-06-27:
As of today's tip of develop, with any program (including tinker) loaded, i see the following behavior:
Following a WiFi disconnect the photon begins flashing blue (as if in listening mode) or sometimes begins flashing green at the exact same speed. However, the photon's SoftAP network is not active, and it was not requested to go into listening mode or whatever green fake listening mode that is. 

Steps to reproduce
1. Load on tinker and today's system firmware from `develop`
2. Give Photon creds for a present WiFi network
3. Let it connect (breathing cyan)
4. Remove the network (i unplugged the router)
5. Observe the photon errantly go into fake listening mode or a slow blinking green. Artifacts are seen in its transition to this state, as if it was confused what state to go into. 

---

Completeness:
- [x] Minimum test case added
- [x] Device, system and user firmware versions stated
- [ ] Particle confirmed

technobly, 2016-06-28:
Hi @pomplesiegel!  I just tried this out at the latest `develop` (https://github.com/spark/firmware/commit/645bb11a20521681b6559a6bf6d96395d767fbcb) running tinker which is definitely further along than when you tested it this morning.  I do not see the same response (which is a good thing!).  I tested by turning off the internet within the router, and also completely powering it off.  With the internet off, it blinked cyan as expected.  With the router off it blinked green as expected.  The photon recovered in both cases when internet/power was restored.

Would you please test again with the latest develop per above and see if you get the same response?

technobly, 2016-06-28:
I also just went back to the way things were this morning (983e25e00bd2b1ed5c2b06a14f6629d563760ced) with the same results.

pomplesiegel, 2016-06-28:
Interesting. This is reproducible on a completely different photon in a different location w/ the following steps: 

1. Load `Tinker` + System fw compiled from today’s develop
2. Clear WiFI creds by holding setup button.
3. Give wifi creds for present WPA2 network (in my case, an iPhone) via **SERIAL** command “w”
4. Allow photon to connect (breathing cyan)
5. Remove network (in my case, turn off the personal hotspot my iPhone is transmitting)
6. Observe weird slow blinking green or blue, even though it's not transmitting a SoftAP network. This speed of green is not an actual state, right?

Feel free to PM me on slack if we can go through this together live. Also, sometimes it only happens on the second or third try, so give it a few shots.

pomplesiegel, 2016-06-28:
OK: more data. This is easiest to reproduce using an iPhone as a wireless hotspot. We just did so using an iPhone 6 and 5SE. Perhaps there's something about the way an iPhone configures itself as a router. Either way, this should now be easy to reproduce. Let me know if you have any further questions!

pomplesiegel, 2016-06-28:
Also, this just occurred at startup on a photon looking for our non-present iPhone hotspot network.

technobly, 2016-06-28:
Interesting, I'll give that a shot. I had used my ASUS router for the test last night.

pomplesiegel, 2016-06-28:
Great, looking forward to it!

technobly, 2016-06-28:
Alright, I tried this on an iPhone 6S Plus and iPhone 5S.  Between the two phones I only ever made it drop to slow blinking blue once on the 5S, after about 5 tries.  Then have not seen it again for many more tries.  Another funny thing I saw was when turning off/on the hotspot connection, several times (3) I saw blinking cyan and orange blips continuously.  It would not recover until I pressed reset after a couple minutes.  Another time I did see it recover from this state, so it might have just needed longer.  Every time the hotspot is turned off, there is a good chance that you'll see the RGB led blink blue for one blip before it starts blinking green.

I did now just see the slow blinking green as well.  It's not in Soft AP mode, or available via USB in Listening Mode.

Now I'd like to know if the same thing happens with 0.5.1.

pomplesiegel, 2016-06-28:
interesting. the issue seems to be made worse by running a real program (aka. not tinker) in the background. I'm seeing it pop up by accident when going about normal business. It's about 75% success rate for us, even when running Tinker. We see both 

1. blue flashing (fake listening mode, as the SoftAP network is not present), and 
2. Slow blinking green (same speed as listening mode blink)
3. w/ our program running, alternating between 2 (above) and normal blinking green

pomplesiegel, 2016-06-28:
@technobly, this is not reproducible using 0.5.1 and Tinker, so the issue must have been introduced following that release.

technobly, 2016-06-29:
Just an update, I have a fix for this that I will submit in the morning ;-)

pomplesiegel, 2016-06-29:
Great! Looking forward to hearing what caused this.

technobly, 2016-06-29:
@pomplesiegel please give this PR a whirl and let me know if the issue is resolved for you: https://github.com/spark/firmware/pull/1050

pomplesiegel, 2016-06-29:
Cool, i'll give that a shot and report back. Are we seeing this because `SYSTEM_FLAG_RESET_NETWORK_ON_CLOUD_ERRORS` is now selectable?

technobly, 2016-06-29:
It was mainly because in adding `SYSTEM_FLAG_RESET_NETWORK_ON_CLOUD_ERRORS` to the `handle_cfod()` function, and refactoring it, one of the reset paths was removed.  Also, `handle_cloud_errors()` was never very effective in the past (which shows the slow blinking orange LED).  When SYSTEM_FLAG_RESET_NETWORK_ON_CLOUD_ERRORS is disabled, it was also effectively using the slow blinking `handle_cloud_errors()` as a blue/green LED instead of orange/red... and getting stuck there indefinitely.

pomplesiegel, 2016-06-30:
just tested - issue still persists and is easily reproducible w/ this PR combined w/ #1047

pomplesiegel, 2016-06-30:
update: Instantly reproducible every time w/ Tinker and the exact same iPhone hotspot procedure described above. Did you try this out locally? I'm seeing no change or improvement at all.

pomplesiegel, 2016-06-30:
Just tried w/ only PR #1050 (rather than a merge w/ #1047) and the issue still persists. it behaves identically to before.

pomplesiegel, 2016-07-01:
@technobly, this bug just occurred w/ a Photon set up w/ a completely normal router, when disconnecting due to a lack of solid connection. We didn't have to use an iPhone or attempt to induce the bug. It occurred naturally. This one is stuck on the slow flashing green.

Have you reproduced this yet w/ an iPhone? If not, how can we help you get more info to solve this? It would be quite unfortunate for this new bug to be present in 0.5.2. 

it's stuck on slow flashing green.

technobly, 2016-07-01:
@pomplesiegel I found the cause of why I thought PR #1050 fixed things for me and wasn't for you.  I was testing with Tinker modified with logging enabled and it worked great.  When I disabled logging I was seeing the issue again.  It turns out there are multiple things fixed with #1050, however the LED signaling is still an issue.  I applied some fixes to that again and it's working 100% with unmodified Tinker.  I'm not happy with these changes though, since they block the real time operation of the system when the error code is generated (most noticeable when running in multi-threaded mode).  We'll work on moving them into the LED signaling interrupt handler for v0.6.0-rc.1, possibly as a separate PR based on #1050.

>It would be quite unfortunate for this new bug to be present in 0.5.2.
I have not seen this on 0.5.2-rc.1 and unmodified Tinker .. have you?  AFAIK this was only present due to some changes recently made in preparation for 0.6.0-rc.1.

pomplesiegel, 2016-07-05:
@technobly, the move into the LED signaling interrupt handler sounds like a good plan to avoid small patches which may be a moving target. Thanks!

My mistake - i assumed these changes went into 0.5.2 as well. Thanks for the explanation.

PR #1050 (2016-07-20): replaces lost reset logic from 'feature/reset-network-flag' (fixes #1045)
- replaces lost reset logic from 'feature/reset-network-flag' (fixes #1045)
- also adds CLR_WLAN_WD() when SYSTEM_FLAG_RESET_NETWORK_ON_CLOUD_ERRORS is disabled
- handle_cloud_errors() was not displaying orange blinks due to HAL_Delay_Milliseconds() not blocking
- handle_cloud_errors() was moved to the end of establish_cloud_connection() to ensure errors are shown
- added debugging info for all of the places ARM_WLAN_WD/CLR_WLAN_WD is called
- added new logic for notify_connected() to ensure ARM_WLAN_WD is not called aft [... cut]
commit a06a3c1c (2016-07-20): replaces lost reset logic from 'feature/reset-network-flag', fixes #1045

- also adds CLR_WLAN_WD() when SYSTEM_FLAG_RESET_NETWORK_ON_CLOUD_ERRORS is disabled
- handle_cloud_errors() was not displaying orange blinks due to HAL_Delay_Milliseconds() not blocking
- handle_cloud_errors() was moved to the end of establish_cloud_connection() to ensure errors are shown
- added debugging info for all of the places ARM_WLAN_WD/CLR_WLAN_WD is called
- added new logic for notify_connected() to ensure ARM_WLAN_WD is not called after DHCP resolves successfully
commit b4067242 (2016-07-20): Merge pull request #1050 from spark/feature/reset-network-flag-fixes

replaces lost reset logic from 'feature/reset-network-flag' (fixes #1045)
closed 2016-07-20 by technobly