particle-iot/device-os #1054: Failing to build the firmware with -werror

ricklesauceur, 2016-07-03:
On a new setup:
Win10 with the latest tool chain / source.

Failing to build the firmware from scratch with -werror. Here we go, error: "BYTE_ORDER" redefined :

`In file included from ./src/photon/wiced/network/LwIP/WWD/FreeRTOS/arch/cc.h:69:0,
                 from ./src/photon/wiced/network/LwIP/ver1.4.0.rc1/src/include/lwip/arch.h:43,
                 from ./src/photon/wiced/network/LwIP/ver1.4.0.rc1/src/include/lwip/debug.h:35,
                 from ./src/photon/wiced/network/LwIP/ver1.4.0.rc1/src/include/lwip/opt.h:46,
                 from ./src/photon/wiced/network/LwIP/ver1.4.0.rc1/src/include/ipv4/lwip/ip_addr.h:35,
                 from ./src/photon/wiced/network/LwIP/WICED/wiced_network.h:43,
                 from ./src/photon/include/wiced_tcpip.h:44,
                 from ./src/photon/include/wiced.h:46,
                 from src/photon/core_hal.c:28:
./src/photon/wiced/network/LwIP/WWD/FreeRTOS/cpu.h:43:0: error: "BYTE_ORDER" redefined [-Werror]
 #define BYTE_ORDER LITTLE_ENDIAN
 ^
In file included from c:\program files (x86)\gnu tools arm embedded\5.4 2016q2\arm-none-eabi\include\sys\types.h:67:0,
                 from c:\program files (x86)\gnu tools arm embedded\5.4 2016q2\arm-none-eabi\include\stdio.h:48,
                 from ./src/photon/wiced/WWD/include/wwd_debug.h:40,
                 from ./src/photon/wiced/WWD/include/wwd_assert.h:42,
                 from ./src/photon/wiced/RTOS/FreeRTOS/WWD/ARM_CM3/FreeRTOSConfig.h:41,
                 from ./src/photon/wiced/RTOS/FreeRTOS/ver7.5.2/Source/include/FreeRTOS.h:78,
                 from ./src/photon/wiced/RTOS/FreeRTOS/WWD/wwd_rtos.h:46,
                 from ./src/photon/wiced/WWD/include/RTOS/wwd_rtos_interface.h:47,
                 from ./src/photon/wiced/platform/MCU/STM32F2xx/peripherals/platform_mcu_peripheral.h:56,
                 from ./src/photon/wiced/platform/include/platform_peripheral.h:42,
                 from ./src/photon/include/wiced_platform.h:46,
                 from ./src/photon/include/wiced.h:44,
                 from src/photon/core_hal.c:28:
c:\program files (x86)\gnu tools arm embedded\5.4 2016q2\arm-none-eabi\include\machine\endian.h:20:0: note: this is the location of the previous definition
 #define BYTE_ORDER _BYTE_ORDER
 ^
cc1.exe: all warnings being treated as errors
make[2]: *** [../build/target/hal/platform-6-m/./src/photon/core_hal.o] Error 1
make[2]: Leaving directory `/c/dev/photon/firmware/hal'
make[1]: *** [hal] Error 2
make[1]: Leaving directory `/c/dev/photon/firmware/modules/photon/system-part1'
make: *** [/c/dev/photon/firmware/modules/photon/system-part1/makefile] Error 2`

m-mcgowan, 2016-07-03:
Thanks for the issue report. I imagine this is because you are using arm-gcc 5.4. Please use an earlier version of arm-none-eabi - 5.4 has only just been released and clearly breaks some things in our toolchain.

ricklesauceur, 2016-07-03:
It works by downgrading to 5.3 . Thanks.

Babbleshack, 2016-08-06:
This issue persists for me, even after downgrading. 

I am using Arch Linux. 
I have tried with toolchains 5.1/5.3/6.1.1 and the latest source.

jhodapp, 2016-09-10:
Any chance of getting this fixed with gcc-5.4? I use the PPA from the linaro gcc Launchpad page and there isn't an older build previous to gcc-5.4. How often does the Particle test to make sure their firmware is up to date with the latest gcc?

PR #1159 (2016-11-29): Compilation fixes for GCC 5.4.x
Closes #1085 and #1054. Tested with GCC 5.4.1, 5.3.1 and 4.9.3

---

Doneness:

- [X] Contributor has signed CLA
- [X] Problem and Solution clearly stated
- [x] Code peer reviewed
- [x] API tests compiled
- [ ] Run unit/integration/application tests on device
- [x] Add documentation
- [ ] Add to CHANGELOG.md after merging (add links to docs and issues)

### Enhancements

- Added support for GCC 5.4.x [[Closes #1085]](https://github.com/spark/firmware/issues/1085) [[Closes #1054]](https://github.com/spark/firmware/issues/1054)
closed 2016-11-29 by technobly