particle-iot/device-os #1060: Build firmware failed  at linking stage

XuGuohui, 2016-07-12:
Hi,

I pulled the latest develop branch and built it with ARM GCC 4.9.3. 

Make command : `PLATFORM=photon PARTICLE_DEVELOP=1`

Error reported:

> src/main.c: In function 'main':
MCU/STM32F2xx/STM32_USB_OTG_Driver/src/usb_core.c:300:27: error: iteration 6 invokes undefined behavior [-Werror=aggressive-loop-optimizations]
MCU/STM32F2xx/STM32_USB_OTG_Driver/src/usb_core.c:298:3: note: containing loop
MCU/STM32F2xx/STM32_USB_OTG_Driver/src/usb_core.c:306:25: error: iteration 6 invokes undefined behavior [-Werror=aggressive-loop-optimizations]
MCU/STM32F2xx/STM32_USB_OTG_Driver/src/usb_core.c:304:3: note: containing loop
lto1.exe: all warnings being treated as errors

Any idea would be appreciated!

XuGuohui, 2016-07-25:
The issue persists until commit cd2f11bc5e57b0c5c1b26e54626376063c986f66

XuGuohui, 2016-08-03:
After digging the source code, I probably found the problem:

In `platform/MCU/STM32F2xx/STM32_USB_OTG_Driver/src/usb_core.c`:

```
for (i = 0; i < pdev->cfg.host_channels; i++)
  {
    pdev->regs.HC_REGS[i] = (USB_OTG_HC_REGS *)(baseAddress + \
      USB_OTG_HOST_CHAN_REGS_OFFSET + \
        (i * USB_OTG_CHAN_REGS_OFFSET));
  }
  for (i = 0; i < pdev->cfg.host_channels; i++)
  {
    pdev->regs.DFIFO[i] = (uint32_t *)(baseAddress + USB_OTG_DATA_FIFO_OFFSET +\
      (i * USB_OTG_DATA_FIFO_SIZE));
  }
```

The `host_channels` may up to 12 in the context, while both of the `HC_REGS[]` and `DFIFO[]` are limited to `USB_OTG_MAX_TX_FIFOS` (4 or 6) elements. The `USB_OTG_MAX_TX_FIFOS` used to be 15 in `platform/MCU/STM32F2xx/STM32_USB_OTG_Driver/inc/usb_core.h`. After certain commit, the `USB_OTG_MAX_TX_FIFOS` is moved to `platform/MCU/STM32F2xx/SPARK_Firmware_Driver/inc/platform_config.h` and is set to 4 or 6 conditionally. 

Any enhancement is planed to fix this issue?

Regards,
guohui

sergeuz, 2016-08-03:
I usually see this obscure error message when I'm trying (accidentally) to build the firmware outside of the `modules` or `main` directories. Could that be your case too?

XuGuohui, 2016-08-04:
@sergeuz Bingo! I did try building the firmware from the `main` and `module` directories and no error occurred. So it is the problem when building the firmware  from root directory, i.e. building the bootloader. 

Would it be fixed soon? I'm eager to port the latest firmware for RedBear Duo.

Thank you so much!

XuGuohui, 2016-08-04:
When linking the bootloader, the `-flto` option is applied, while building the modular firmware and user part dose not. That's why building from the `module` and `main` directories didn't launch this error.

jenesaisdiq, 2016-08-15:
Hey @XuGuohui, is this resolved? Want to make sure you've got the Duo happily running Particle!

XuGuohui, 2016-08-16:
The target issue had been figured out on post #3. I can temporarily modify the `USB_OTG_MAX_TX_FIFOS` value to get it compiled through. But I'd suggest Particle to have a commit to fix this issue.

PR #1179 (2016-11-29): Feature/bootloader fixes
- Builds bootloader on Travis
- Fixes #1178 on Electron
- Fixes #1060

---

Doneness:

- [X] Contributor has signed CLA
- [X] Problem and Solution clearly stated
- [x] Code peer reviewed
- [x] API tests compiled
- [x] Run unit/integration/application tests on device (N/A)
- [x] Add documentation (N/A)
- [ ] Add to CHANGELOG.md after merging (add links to docs and issues)

### Bugfixes

- [Electron] Bootloader build was failing, fixed and added to CI. [[Fixes #1178]](https://github.com/spark/firmware/issues/1178) [[Fixes #1060]](https://github.com/spark/firmware/issues/ [... cut]
closed 2016-11-29 by technobly