particle-iot/device-os #1067: as a user, I can interrupt the blinking cyan cloud connection with the setup button

m-mcgowan, 2016-07-25:
At present, pressing the setup button to enter listening mode, or a double tap to powerdown, doesn't interrupt the cloud connection process. (It does interrupt the network connection process.)
---

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

technobly, 2016-08-01:
Related to https://github.com/spark/firmware/issues/1013

technobly, 2016-09-09:
@m-mcgowan just to clarify, does this apply to single-threaded mode as well... or just multi-threaded mode?

m-mcgowan, 2016-09-13:
For both!

sergeuz, 2017-02-05:
Let's put this on hold for now, since in case of Photon, there isn't an easy way to break a blocking call to the socket API other than shutting down WICED connectivity layer (and thus breaking lwIP processing loop), which is buggy in the WICED release that we currently use.

technobly, 2017-02-07:
@sergeuz This would still be very useful for the Electron, which can sometimes have a long connection time, especially if the unit is encountering a handshaking issue.  The implementation for Electron could be stubbed out for Photon/P1 and revisited after WICED is updated.

PR #1260 (2017-05-04): Interrupt cloud connection when entering listening mode or powering off the device via setup button
This PR fixes #1067 by introducing a flag that can be set from an ISR and cause read/write operations on the cloud socket to fail with a generic error, effectively breaking any busy loop within the comms library.

---

Doneness:

- [X] Contributor has signed CLA
- [X] Problem and Solution clearly stated
- [ ] Code peer reviewed
- [ ] API tests compiled
- [ ] Run unit/integration/application tests on device
- [ ] Add documentation
- [ ] Add to CHANGELOG.md after merging (add links to docs and iss [... cut]
closed 2017-05-04 by technobly