particle-iot/device-os #1075: Calling `System.sleep(SLEEP_MODE_DEEP, sleepForSeconds)`, while the Cellular modem is active, with a small value of `sleepForSeconds` never wakes up again

RobMeades, 2016-08-02:
I have several 10's of devices containing Particle Electron boards (Sara-U270 ones).  They were all working perfectly until I restructured the code to call `System.sleep(SLEEP_MODE_DEEP, sleepForSeconds)`, rather than `System.sleep(WKP, RISING, sleepForSeconds, SLEEP_NETWORK_STANDBY)`.  The problem scenario is as follows (see thread here: https://community.particle.io/t/particle-electron-sleep-duration-limitations-are-there-any/24688):

1. Perform a connect/publish action with the modem, so the modem is awake.
2. Call `System.sleep(SLEEP_MODE_DEEP, 30)`.
3. System never wakes up.

I have a separate interrupt running to the `WKP` pin from an accelerometer and even that going off fails to cause a wake-up in this fail case. The failure does not occur all the time but it does eventually occur.  It does not appear to occur with longer wake-up delays. If I add a step between 1 and 2:

1a. Call `Cellular.off()`.

...then, so far at least, the problem goes away.  I will update this issue if I get any more information during testing.

This is with system firmware `0.5.1`.  The full code is available here: https://github.com/u-blox/tracker/blob/master/tracker.cpp.

---

Completeness:
- [ ] Minimum test case added
- [x] Device, system and user firmware versions stated
- [ ] Particle confirmed

avtolstoy, 2016-08-02:
This should be fixed in PR #1051 (see original issue #1043). It's already merged into 0.6.x and will also be backported into 0.5.x.

RobMeades, 2016-08-02:
@avtolstoy: ah, good.  I can't wait for the fix unfortunately as this code is meant to have been delivered already.  Can you confirm that calling `Cellular.off()` before the deep sleep call is a good workaround or, if not, what would be a good workaround?

avtolstoy, 2016-08-02:
@RobMeades I think something like this should work in theory:
```c++
Cellular.disconnect();
Cellular.off();
while(Cellular.ready()) {
  Particle.process(); // or delay(1); maybe
}
System.sleep(SLEEP_MODE_DEEP, 30);
```

Please also read the issue #1043, there were a couple of workarounds mentioned there, but seems that simply calling `network_disconnect/network_off` (or `Celullar.disconnect()/Cellular.off()`) didn't work. I believe that waiting for `Cellular.ready()` to go `false` should ensure that the modem was disabled.

RobMeades, 2016-08-02:
@avtolstoy: may thanks, I'll try that.

technobly, 2016-08-14:
currently in v0.6.0-rc.1 and v0.5.3-rc.2

closed 2016-08-14 by technobly