particle-iot/device-os #1077: gcc-arm-none-eabi-49 is no longer available from homebrew; using gcc-arm-none-eabi

vielmetti, 2016-08-02:
The build instructions specify prerequisites for Mac OS X, that include installing `gcc-arm-none-eabi-49` via a Homebrew tap. That version is no longer labelled as such, but it's possible to install `gcc-arm-none-eabi`.

When I do that, I get this version string:

```
Edwards-MacBook-Air:firmware emv$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]
```
which I think is correct.

---

Completeness:
- [x] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

technobly, 2016-08-02:
We are currently using this specific build of version 4.9.3 internally:
```
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150303 (release) [ARM/embedded-4_9-branch revision 221220]
```

vielmetti, 2016-08-02:
The exact 4.9.3 version that Homebrew picks up is at

https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q3-update/+download/gcc-arm-none-eabi-4_9-2015q3-20150921-mac.tar.bz2

There's a full set of [release notes](https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q3-update/+download/release.txt) describing what's different.  I see several small bug fixes (er, small if they are not your bugs, large if they are your bugs).

vielmetti, 2016-08-02:
The Homebrew tap pulls the files from

https://launchpad.net/gcc-arm-embedded/4.9

which looks like it has a full set of details, should someone need to pull a particular version.

technobly, 2016-08-02:
We should probably fix the instructions to represent the same build version we are using internally for now.  Not sure if this means we need to create a tap of our own that we can better control when it updates, but something like that. I don't think it hurts to have the latest 4.9.3 but to stay in sync with Particle, 4.9.3 20150303 is the current build we are using in the Cloud.

technobly, 2016-08-02:
Here's the formula that should get you the same version:
https://github.com/PX4/homebrew-px4/blob/aec23308d7df9f5c19b2245a5e8c9dbd2b62c041/gcc-arm-none-eabi-49.rb

stoto, 2016-08-23:
I am having the same problem. How can i install this formula?

BracWebb, 2016-08-23:
FYI -- I have built with `px4/px4/gcc-arm-none-eabi-54` installed through homebrew and it seems to work fine

stoto, 2016-08-24:
@BracWebb I tried it as well but i am running into these issues:

`opt/local/arm-none-eabi/include/sys/time.h:38:19: error: field 'it_interval' has incomplete type
   struct  timeval it_interval;
                   ^
/opt/local/arm-none-eabi/include/sys/time.h:39:19: error: field 'it_value' has incomplete type
   struct  timeval it_value;
                   ^
lib/mbedtls/library/timing.c:66:20: error: field 'start' has incomplete type
     struct timeval start;
                    ^
lib/mbedtls/library/timing.c: In function 'mbedtls_timing_get_timer':
lib/mbedtls/library/timing.c:285:20: error: storage size of 'offset' isn't known
     struct timeval offset;
                    ^
lib/mbedtls/library/timing.c:285:20: error: unused variable 'offset' [-Werror=unused-variable]`

stoto, 2016-08-24:
I managed to install 4.9.3 and it compiles now, thanks.
That article helped me do just that: http://mygeekdaddy.net/2014/12/05/how-to-install-a-local-file-in-homebrew/

thelostspore, 2016-09-07:
This downloads the formula posted above by @technobly & puts it into your local `px4` tap:
```bash
curl -s https://raw.githubusercontent.com/PX4/homebrew-px4/\
aec23308d7df9f5c19b2245a5e8c9dbd2b62c041/\
gcc-arm-none-eabi-49.rb > \
/usr/local/Library/Taps/px4/homebrew-px4/gcc-arm-none-eabi-49.rb
```
 Note: `brew install gcc-arm-none-eabi-49` does emit some Formulae deprecation warnings, but this appears to be working.

technobly, 2016-11-25:
@thelostspore thanks for posting your tips.  Unfortunately for me when I tried to follow it, homebrew didn't like the `sha1` included in the old formula and wanted a `sha256` to proceed.  I've edited the formula and tested the install.  Here are my steps:

### 1) Copy/Paste this in Terminal and press ENTER to create the proper Brew formula
```
echo -e "require 'formula'

class GccArmNoneEabi49 < Formula
  homepage 'https://launchpad.net/gcc-arm-embdded'
  version '20150306'
  url 'https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q1-update/+download/gcc-arm-none-eabi-4_9-2015q1-20150306-mac.tar.bz2'
  sha256 '049469f06c35d1a7c6d333236bc62ed340531c4377b4e16ce8c5aa8d134e8e8b'

  def install
    ohai 'Copying binaries...'
    system 'cp', '-rv', 'arm-none-eabi', 'bin', 'lib', 'share', \"#{prefix}/\"
  end
end" > /usr/local/Homebrew/Library/Taps/px4/homebrew-px4/gcc-arm-none-eabi-49.rb
```

### 2) install it!
    brew install gcc-arm-none-eabi-49

closed 2016-12-08 by technobly

beadon-SAT, 2020-01-26:
better, do this : brew cask install gcc-arm-embedded

PR #1183 (2016-12-08): fixes docs to install arm-none-eabi-gcc version 4.9.3 20150303
The `homebrew-px4` tap keeps updating the latest GCC compiler version and does not have `gcc-arm-non-eabi-49` any longer.  This edit to the documentation preserves the info needed to install v4.9.3 20150303 of arm-none-eabi-gcc which is the recommended version to use locally when building firmware.

Fixes #1077 
Closes #1103 #1078

---

Doneness:

- [x] Contributor has signed CLA
- [x] Problem and Solution clearly stated
- [x] Code peer reviewed
- [x] API tests compiled (N/A)
- [x] Run unit/integration/application tests on device ( [... cut]