particle-iot/device-os #1080: SPI1 DMA fails for Electron for 0.5.1 and 0.5.3rc1

greiman, 2016-08-04:
Electron SPI1 DMA fails with a red SOS error for the example code below.  The code works for Electron SPI DMA.

Here are some test results:

> This code works on a Photon (0.5.2)
> It causes an SOS fault on the Electron running 0.5.3rc1
> It causes an SOS fault on the Electron running 0.5.1
> It causes an SOS fault on the Electron running 0.5.0
> It works on the Electron running 0.4.8.

```
#include "application.h"

SYSTEM_MODE(MANUAL);
uint8_t buf[512];

volatile bool spiDone;

void spiCallback() {
  spiDone = true;
}

void setup() {
  Serial.begin(9600);
  SPI1.begin();
}

void loop() {
  spiDone = false;
  Serial.println("read");
  SPI1.transfer(0, buf, sizeof(buf), spiCallback);
  while (!spiDone) {
    delay(1);
  }

  Serial.println("write");
  spiDone = false;

  SPI1.transfer(buf, 0, sizeof(buf), spiCallback);
  while (!spiDone) {
    delay(1);
  }
  Serial.println("done");  
  delay(1000);
}
```

---

Completeness:
- [X ] Minimum test case added
- [X ] Device, system and user firmware versions stated
- [ ] Particle confirmed

rickkas7, 2016-08-04:
More in this thread:
https://community.particle.io/t/electron-spi1-dma-fails-in-0-5-1/24874

I was able to reproduce SPI1 working in 0.4.8 with that code and not working in later versions.

ScruffR, 2016-08-05:
@rickkas7, in your post in that thread, you stated that 0.5.2 was OK too?

avtolstoy, 2016-08-05:
This is a bug. `DMA1_Stream2` entry in the vector table was not pointing to `DMA1_Stream2_irq()` defined in `spi_hal.c`. This is fixed in PRs #1082 (0.5.x) and #1081 (0.6.x).

These PRs will not make it into `0.5.3-rc.2` and `0.6.0-rc.1` and will only be included in the next release.

rickkas7, 2016-08-05:
This issue was the simple fail case for the sdfat library not working on the Electron when used with SPI1. I tested 0.5.3rc2 with PR  #1082 applied and the sdfat TryMeFirst sample app now works properly with SPI1 as well. Looks good, thanks!

closed 2016-09-01 by technobly