particle-iot/device-os #1084: Compilation error using ARM GCC V. 6.1.1. 20160526

jhlink, 2016-08-09:
< Please summarize your issue here.  Check [x] as many boxes below as you can before submitting your issue.  Thanks -- you rock! >

---

I'm running 64-bit Arch Linux with kernel version 4.6.4-1 with the following ARM GCC tool chain version from the arch official repository: arm-none-eabi-gcc (Arch Repository) 6.1.1 20160526

On a clean fork of the 'latest' branch, I found the following errors. 

Executing

`make clean all PLATFORM=photon` within `$firmware/modules` results with various `-Werror` compilation errors. 

Executing the same the command within `$firmware/main` results with the following error
`In file included from ../wiring/inc/spark_wiring_thread.h:31:0,
                 from ../wiring/inc/spark_wiring_watchdog.h:22,
                 from ../wiring/inc/spark_wiring_cloud.h:31,
                 from ../wiring/inc/spark_wiring.h:48,
                 from ./inc/application.h:36,
                 from src/application.cpp:21:`

`/usr/arm-none-eabi/include/c++/6.1.1/mutex: In member function 'bool std::recursive_timed_mutex::_M_timedlock(const __gthread_time_t&)':
/usr/arm-none-eabi/include/c++/6.1.1/mutex:290:69: error: '__gthread_recursive_mutex_timedlock' was not declared in this scope
       { return !__gthread_recursive_mutex_timedlock(&_M_mutex, &__ts); }
                                                                     ^
make[2]: *** [../build/module.mk:235: ../build/target/user/platform-6-m/src/src/application.o] Error 1
make[2]: Leaving directory '/home/jhkim012/GitRepo/firmware/user'
make[1]: *** [../../../build/recurse.mk:12: user] Error 2
make[1]: Leaving directory '/home/jhkim012/GitRepo/firmware/modules/photon/user-part'
make: *** [../build/recurse.mk:12: modules/photon/user-part] Error 2
`

ARM GCC v5.4.1 20160609 is reported to return different compile errors as well. 

Currently, the temporary solution is to revert to v4.9.3.

Completeness:
- [x] Minimum test case added
- [x] Device, system and user firmware versions stated
- [x] Particle confirmed

sergeuz, 2016-11-12:
I don't see GCC 6.x available for download [here](https://launchpad.net/gcc-arm-embedded), on the page where ARM usually publishes their compiler binaries, and Arch package mentioned in the issue seems to be provided by a community: https://www.archlinux.org/packages/?sort=&q=arm-none-eabi-gcc&maintainer=&flagged=

I would wait until ARM makes stable release before looking into any compilation issues related to GCC 6.x series. Meanwhile, feel free to check out fixes for GCC 5.4.x: https://github.com/spark/firmware/pull/1159

mathben, 2016-11-14:
I have the same error and environment of jhlink.
Arch Linux
ARM GCC 6.2 from repo

I checkout feature/build_fixes and it's fix warning.

I would like to fix it, but I don't think the problem can be solve from spark/firmware

I would like to test with https://launchpad.net/gcc-arm-embedded/+milestone/6-2016-q4-major
But I don't find how to download it for test.

jhlink, 2016-11-22:
Hey @sergeuz and @mathben 

I talked to the Particle team about this a few months back. They don't have support for the latest versions of ARM GCC (v6.2.x), but they recommended using an older version of the toolchain, specifically v4.9.x

This will solve all these weird problems. 

Tackling all these warnings and all that jazz is going to be quite a task... I've attempted it, but warnings and errors continue to pop up. Even after setting the "ignore warnings" flag (I was feelin' brave here) and rewriting some lines of code.

m-mcgowan, 2016-12-14:
> EDIT: Can't wait for support for the later toolchain versions!

Yeah, always good with new things :-) What features are there in the new tools that you need? Always good to know how folks are using toolchains and writing code :-)

sergeuz, 2017-01-24:
Related PR: https://github.com/spark/firmware/pull/1218

technobly, 2017-02-09:
0.7.0-rc.1 will be targeting GCC 5.3.1

closed 2020-02-29 by avtolstoy