particle-iot/device-os #1086: Expose thread id to user app

hossboss, 2016-08-10:
I have a few cases where for concurrent purposes I need to identify which thread a function is running on. There are a few `os_thread_` and `__gthread_` functions that compare thread ids, but `__gthread_self()` is not exposed, so I cannot just grab the id. There are a few workarounds I've been considering, but they are not nearly as clean (or fast, or low overhead). 

What is the possibility of exposing this function?

monkbroc, 2016-08-23:
The API for managing user created threads is minimal right now (mostly on the [Thread class](https://github.com/spark/firmware/blob/138e22d64c2d4f1102b165bc4387aa5f5319bbb2/wiring/inc/spark_wiring_thread.h#L37)). It's something we can improve. We're definitely open for suggestions!

hossboss, 2016-08-25:
My goal for being able to identify a thread is to be able to have some sort of thread-local storage. I was looking for it for an experimental library I am working on for exception handling. In this case there is code that could be invoked by any thread that needs to be able to maintain a separate state for each thread. As a workaround I set up a method of storing the thread ids when creating a thread, and then iterating the list and calling `os_thread_is_current()` to find the current thread and local storage index.

A general solution for thread-local storage would be pretty cool. Something like the C++ `thread_local` or GCC `__thread` storage classes would be ideal and very easy to use...but looks to need quite a bit of extra machinery to light up, if even possible.

I found a workaround for my particular use case. I have no idea if this is something anyone else would want.

monkbroc, 2016-08-26:
Thanks for the explanation. It's quite specific as a use case for sure.