particle-iot/device-os #1087: Photon: I2C slave can not handle restart operation

XuGuohui, 2016-08-10:
Hi there,

According to the I2C hal source code, the Photon now acting as I2C slave is not capable of handling restart data transmission. The onReceive callback is invoked only if a stop condition is detected, despite of a restart condition.

avtolstoy, 2016-10-30:
@XuGuohui You are correct. I've just taken a look into this and it is not possible to detect a _repeated-START_ (or just a _START_ for that matter). STM32F2 I2C peripheral only generates an interrupt when a configured address is matched. This is not very useful, since if a Master generates a _repeated-START_ and talks to another device on the bus, our Slave will not see the end of the transaction at all.

We should mention in the documentation that when configured as a Slave, all transactions should be terminated with a STOP condition and that repeated START is not supported.

XuGuohui, 2016-10-31:
Thank you for your clarification.

closed 2016-10-31 by XuGuohui