particle-iot/device-os #1093: [0.6.0-rc.1] USB HID should not need Windows driver

hramadan, 2016-08-14:
It seems that under Windows, the USB HID device is not exposed to the O/S, but Device "Photon" is shown in Device Manager.

Using Photon's USB HID interface should *not* rely on having a Windows driver installed, eg like Teensy https://www.pjrc.com/teensy/ 

Refer: https://community.particle.io/t/0-6-0-rc-1-usb-hid-print-bug-ms-windows-issue/25129. I note that tests were conducted under a VM and not real hardware (this will have to wait until the end of August 2016....).

void setup() {
      Keyboard.begin();
}
void loop() {
      // Should not need a Windows driver installed to operate
      Keyboard.write("A");
      delay(5000L);
}

System firmware 0.6.0.rc1, PHOTON

---

Completeness:
- [X] Minimum test case added
- [X] Device, system and user firmware versions stated
- [ ] Particle confirmed

avtolstoy, 2016-08-14:
@hramadan HID does not require the Particle driver installed and uses standard hidusb.sys/hidclass.sys on Windows. There is a bug in 0.6.0-rc.1 (#1089) that prevents proper enumeration under Windows, due to multiple configurations (normal and high power). PR #1091 also fixes the issue with multiple consecutive HID reports cancelling each other as described in #1090.

A note from Microsoft blog about multiple configurations:
>USBCCGP will not load on a multi-config device by default because the hub driver doesn't create a "USB\COMPOSITE" PNP ID for a composite device if it has multiple configurations. However, you can write your own INF that matches a device-specific PNP ID to get USBCCGP to load as your device's function driver.

Electron's HID on freshly installed Win7 SP1 + latest updates:
[image]

hramadan, 2016-08-14:
@avtolstoy, phew! Excellent!

I will close the ticket once confirmed (but this will be two weeks away).

technobly, 2016-09-09:
Closing out now that v0.6.0-rc.2 is out :)

closed 2016-09-09 by technobly