particle-iot/device-os #1096: [Enhaxement] USB HID Mouse should support scroll wheel and absolute moveTo()

ScruffR, 2016-08-17:
Nowadays USB mice should definetly also support the scrall wheel (z coordinate) and if possible also implement an absolute mouse movement `moveTo()` like Paul Stoffregen has done on his Teensy boards.

https://www.pjrc.com/teensy/td_mouse.html
https://forum.pjrc.com/attachment.php?attachmentid=942&d=1379523560

avtolstoy, 2016-08-17:
@ScruffR Mouse wheel is already supported, it's the third argument to `move()`: https://github.com/spark/firmware/blob/develop/wiring/inc/spark_wiring_usbmouse.h#L59

In general, `Mouse` has the following capabilities:
- 3 buttons
- X, Y
- Wheel

We could totally make a separate `wheel()` function to simplify the usage.

As for absolute screen coordinates support, perhaps it's a good candidate to be implemented as a library?

technobly, 2016-08-17:
what happens when you call Mouse.move() and just want to move the scroll wheel?  Do you have to remember the last x,y and call as Mouse.move(savedX, SavedY, newWheel)?  Likewise for x,y only?  That would be a nice enhancement to add moveTo() and wheel(), but if added to a library, the library would have to remember the last x,y or wheel to use the existing Mouse.move(x,y,wheel) command.  Is there a way in the HID implementation to just send x,y or wheel commands?  That would avoid needing to know the current x,y to send wheel only, or wheel to send x,y only.  What if you just want to move the mouse on x or y separately as well?  moveX, moveY, moveTo, wheel?

avtolstoy, 2016-08-17:
@technobly No need to save anything. If you only want to move x coordinate by 100, you call `Mouse.move(100, 0, 0);`, if only scroll the wheel `Mouse.move(0, 0, 100);`

ScruffR, 2016-08-17:
@avtolstoy, duh' sorry, for some reason I mistook the third parameter as button click :blush:

ScruffR, 2016-08-18:
An actual absolute mouse HID would not first move the pointer to 0/0 and then move relative.
A HID like this is more like a digitizer where the actual position of the pen on the digitizer surface is directly teanslated to the respective coordinate in the screen.
You just need to provide the scale digitizer (0..65535/0..65535) corresponds to screen coordinates (0..1920/0..1080 or -3600..+3600/-2250..+2250).
The USB HID descriptor (and/or a dedicated setup report) would carry the scale information while the regular USB HID report would only carry the raw coordinates (AFAIK).

closed 2016-09-07 by technobly

xcarcelle, 2020-10-01:
Dear @ScruffR : thanks for this great project, I wanted to try absolute Mouse function from device-ios connected with different devices over BLE. Would it be possible to run device-os on a Adafruit Feather nrf52 board ? Or other hardware ?
Have you tested the absolute move with Android/iOS devices ? Regards

ScruffR, 2020-10-01:
Currently the Particel device OS doesn't even support USB HID on their nRF based boards - let alone BLE HID.

PR #1110 (2016-09-07): Feature/usb hid enhancements
Closes #1096

- USBKeyboard rework. Keycodes moved to https://github.com/spark/firmware/blob/8870980a4bfcaae30823cff27a0cee85ab52d379/wiring/inc/spark_wiring_usbkeyboard_scancode.h
- `Keyboard.write()` uses ASCII map to press appropriate keyboard buttons, see https://github.com/spark/firmware/blob/8870980a4bfcaae30823cff27a0cee85ab52d379/wiring/src/spark_wiring_usbkeyboard.cpp#L33
- `Keyboard.writeKey()`, `Keyboard.press()`, `Keyboard.release()` and `Keyboard.click()` use keycodes and modifier codes defined in `UsbKeyboardScanCode` and `UsbKeyboardModifier`, see ht [... cut]