particle-iot/device-os #1106: SLEEP_MODE_DEEP stops working after several hours

rayterrill, 2016-08-29:
I've got some code that's attempting to check a moisture sensor, post the resuts to ThingSpeak, and go into SLEEP_MODE_DEEP for 60 seconds. That works great initially, but after several hours, the Photon just stops responding/posting. Particle Photon, building with firmware 0.5.2.

Sample code:
`#include "ThingSpeak/ThingSpeak.h"

TCPClient client;

int sensorPin = A0;    // set the input pin for the moisture sensor

void setup() {
   ThingSpeak.begin(client);
}

void loop() {
   int sensorValue = 0;

   Serial.println("Reading sensor value...");
   sensorValue = analogRead(sensorPin);
   Serial.println(sensorValue);
   
   Serial.println("Attempting to publish a reading...");
   ThingSpeak.setField(1,sensorValue);

   ThingSpeak.writeFields(MYCHANNEL, "APIKEY");
   
   //delay(30000);
   //go into deep sleep for 60 seconds
   System.sleep(SLEEP_MODE_DEEP, 60);
}`

Can anyone point me in the right direction as far as troubleshooting? Unfortunately, the device is remote, so troubleshooting is pretty brutal (raises hand for deferred OTA updates, please! :smile: ). Any help would be very much appreciated.

m-mcgowan, 2016-09-13:
Could you try the same app without any of the ThinkSpeak code? It would be good to try to narrow down the cause to either our system firmware or the ThinkSpeak library. Thanks!

m-mcgowan, 2016-09-15:
https://community.particle.io/t/photon-not-waking-up-from-sleep-mode-deep-after-several-hours-solved/25330

Issue marked as resolved - seems to hang in the ThingSpeak library.

closed 2016-09-15 by m-mcgowan