particle-iot/device-os #1109: FATAL: New USB stack crashes Windows 7 with "BAD_POOL_CALLER" error

XuGuohui, 2016-09-05:
Hi there,

The latest firmware built based on the latest commit: 61dbd1d39d143574974ad7f62fc000ea777c48fd: "added 0.6.0-rc.1 Docs links to CHANGELOG.md)" will crash the Windows 7.

1. The old USB stack works well on my Win 7 computer, while the new USB stack will crash the Windows every time once the Photon is plugged in and the computer turned into blue screen with error "BAD_POOL_CALLER, 0x000000C2".

2. But the new USB stack works well on some other Win 7 computers, except mine.

I think there is something wrong with my computer, as well as the new USB stack (maybe some new USB features are not supported on my computer). How can I fix it? I suppose there will be some other guys meet the same  issue using the new USB stack.

Thanks in advance!

avtolstoy, 2016-09-05:
@XuGuohui Whoah, BSOD doesn't sound too good. Could you clarify a couple of things?
1. Are you using USB 3.0 or USB 2.0 port?
2. Do you have old Windows drivers (intended for <0.6.0) or new ones (obtained from https://ci.appveyor.com/api/buildjobs/jcdofeop0vnxea4m/artifacts/windows-device-drivers.zip)?
3. Are you using any of the new features (USBSerial1 / Mouse / Keyboard)?

If you are using new features, I suggest following the steps at https://github.com/avtolstoy/particle-usb-testing#windows to install new drivers.

XuGuohui, 2016-09-05:
@avtolstoy 

1. I was using the USB 2.0 port
2. The issue persisted even if I uninstall and delete the old diver.
3. The Photon just run the default tinker application. It seams that once the USB Serial is up, the Windows crashes.

Thanks!

avtolstoy, 2016-09-05:
@XuGuohui Could you provide me with a minidump of the crash (from `C:\Windows\minidump` folder)? You can send it to my profile email as not to share it publicly.

I can also suggest trying to see if following steps 2 and 5 (deleting old drivers and cleaning usb driver cache) from https://github.com/avtolstoy/particle-usb-testing#windows and rebooting afterwards helps the issue.

This is still probably a bug in the usb controller driver or something similar. See if there are newer drivers available for your chipset/usb controller.

XuGuohui, 2016-09-05:
@avtolstoy I've sent you the dump file, please check it out. And I'll try the methods you suggested. 

Thank you so much!

avtolstoy, 2016-09-05:
@XuGuohui Seems pretty standard weird usbser.sys issue. Nothing interesting in the crash dump. I think you have somewhat old usbser.sys though. Maybe updating Win7 will help? There is convenience rollup update KB3125574 which includes everything up to April 2016. I think KB3020369, KB3102810, KB3138612 are prerequisites for it.

XuGuohui, 2017-02-09:
After struggling to make the Windows Update work, I finally update my Windows 7 to the latest version, and the issue was fixed. :smile:

technobly, 2017-02-09:
Great to hear it!  Assuming that is a reliable fix for you, could you close the issue?

closed 2017-02-09 by XuGuohui

XuGuohui, 2017-02-24:
Hi @technobly @avtolstoy ,

Did you update the Windows driver for Photon? We just released a new system firmware version for RedBear Duo, which is based on 0.6.1-rc2 of Paricle firmware, but the old driver is not valid for the new firmware. Please help!

Kind regards.

avtolstoy, 2017-02-24:
@XuGuohui Are you using https://github.com/spark/windows-device-drivers/releases/download/v6.1.0.68/particle_drivers_6.1.0.68.exe? That's the latest installer, however it only installs drivers for PID=0xc006, 0xc008, 0xc00a. There weren't any changes to the firmware that could affect the previous drivers either.

XuGuohui, 2017-02-24:
@avtolstoy  After deploy the new firmware and uninstall the driver, the Duo showed as:

* Duo Control Interface
* Duo Serial
under the Other Device category.

1) If I don't uninstall the previous driver, the Duo can still be recognized as the old name "Duo with WiFi and BLE", instead of new name "Duo". Does it make sense?

2) Should we need to create a new signed driver for the Duo running new firmware? If I uninstall the old driver and re-install, it failed.

avtolstoy, 2017-02-24:
@XuGuohui 

Just a note, that 0.6.0+ should work on Windows 10 without any custom drivers.

1. If you don't use any of the new functionality (USBSerial1 or HID), it should work normally with pre 0.6.0 drivers.

2. As far as I can see, Duo uses `USBD_PID_CDC=0xC058`, so yes, you need to have a similar entry to https://github.com/spark/windows-device-drivers/blob/master/particle.inf#L73 in the inf file and it should be signed. The installer also takes care of uninstalling the old drivers.

XuGuohui, 2017-02-25:
@avtolstoy  I am not familiar with signing a Windows driver for USB device. What I expect is that users can install the new driver to play with the Duo basing on the new system firmware, including the HID feature. But I don't know how to sign a Windows driver. The previous driver for Duo is created by @m-mcgowan and I really appreciate his help! I leave him a private message on the Particle forum, but it seams like he hasn't read the message yet. While we desire the new driver currently, as we are going to ship out a patch of Duo running the new system firmware. I'm afraid that the users without previous driver installed are not able to play with the Duo.

Much appreciate for your help!