particle-iot/device-os #1112: [0.6.0] I2C_HAL.H - I2C_BUFFER_LENGTH increase

hramadan, 2016-09-07:
Is it possible to increase the buffer length used in I2C requestFrom() so that 64 bytes of data can be read at once? The buffer size is currently 32.

Am having an issue with the NXP PN532 NFC reader IC when there is a  more than 32 bytes of data to be transferred. I have not been able to find a work around. 
---

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

hramadan, 2016-09-10:
A better idea has been suggested by @pkourany - rather than increase the internal buffer to some arbitrary size, let the application specify its own buffers. This is a far better solution.

hramadan, 2016-09-14:
Refer to the following post:

https://community.particle.io/t/i2c-requestfrom-how-to-read-32-bytes/25615

Would like similar lower level control as offered by SoftI2CMaster lib, not a larger buffer per se. Am changing the title accordingly.

worldburger, 2019-08-26:
Has there been a simple implementation to reading buffer lengths longer than 32?

hramadan, 2019-08-26:
@worldburger, to my knowledge this request has not been addressed. Particle documentation, https://docs.particle.io/reference/device-os/firmware/photon/#wire-i2c-, still states a **32 byte buffer**. 

Is your problem in relation to reading NFC cards as this was (still is) my issue.

worldburger, 2019-09-06:
@hramadan No, I've got a usage case with a particulate monitor serving up more than 32 bits.

I recognize your icon! You're active in the Particle thread:
https://community.particle.io/t/solved-i2c-requestfrom-how-to-read-32-bytes/25615/18}

The particle library I'm trying to get to work is the SPS30. I had looked at your solution with SoftI2CMaster/SoftI2CLibrary.  But to be honest, it's a bit over my head (am rather new-ish to coding).  Have you found a more turnkey solution to adjusting the buffer?

hramadan, 2019-09-08:
@worldburger, have you looked at the Sensiron source code on GitHub?

https://github.com/Sensirion/embedded-sps
https://github.com/Sensirion/embedded-common

Am pretty sure that you will be able to solve your issue by implementing their source code.

avtolstoy, 2020-02-29:
Fixed in #2035 #2022

closed 2020-02-29 by avtolstoy