particle-iot/device-os #1115: Calling Keyboard.begin() closes the USB Serial Port

hramadan, 2016-09-09:
Calling Keyboard.begin() closes the usb Serial, as evidenced by screen /dev/cu.usbmodem621 returning back to the operating system.

BUT, I can re-rerun screen /dev/cu.usbmodem621, and it opens okay.

Test code:

void setup() { 
      Serial.begin(9600);
      Keyboard.begin();
      delay(1000L);
      Serial.println("Exiting setup");
}

void loop() {
     delay(500L);
     Serial.println(".");
}

Device: PHOTON, v0.6.0.rc2

---

Completeness:
- [X ] Minimum test case added
- [X] Device, system and user firmware versions stated
- [ ] Particle confirmed

avtolstoy, 2016-09-09:
Copying my answer from original thread https://community.particle.io/t/0-6-0-rc2-keyboard-begin-closes-serial-usb-port/25702/4?u=avtolstoy

This is expected. By the time setup() is called, the device has already been enumerated to the host. By default, only Serial is enabled in order to keep compatibility with "old" Windows drivers.

When Keyboard.begin() is called, the device needs to change its USB descriptors that are presented during enumeration, open additional endpoints etc etc. There is no way to do all that without performing a reattach (i.e. disconnect/reconnect).

If that behavior is undesirable, you can always call Keyboard.begin() in STARTUP(), which will make the device attach for the first time after boot with both Serial and Keyboard:

```
STARTUP(Keyboard.begin());

void setup() { 
Serial.begin(9600);
delay(1000L);
Serial.println("Exiting setup");
}

void loop() {
delay(500L);
Serial.println(".");
}
```

hramadan, 2016-09-09:
Confirming Case Closed on this issue (with thanks) as per https://community.particle.io/t/0-6-0-rc2-keyboard-begin-closes-serial-usb-port/25702/5?u=umd

technobly, 2016-09-09:
Thanks guys for closing the loop on this fast!

closed 2016-09-09 by technobly