particle-iot/device-os #1118: Type-safe times in our firmware api

m-mcgowan, 2016-09-13:
Some times are in seconds, while others are in milliseconds. This can cause confusion.

Some possible solutions:

- change APIs to consistently use the same time unit.
- provide a type-safe API that forces the user to specify the time unit.

technobly, 2017-02-07:
The first one seems like a good solution, and you would go with the LCD (lowest common denominator) so out of all API's if milliseconds is the smallest unit of time, then all API's should use milliseconds.  This would definitely present it's own problems when users don't know about the change, and use 10 seconds to deep sleep, instead of 10000 milliseconds.  Unfortunately with the sleep API, the RTC doesn't have millisecond resolution so we'd have to take a value of 10000 and convert it to 10 anyway.  If a user specified 10 on accident after the change, it would basically result in sleep for 0 seconds.  If microseconds is used in any API, things could get pretty ugly fast (and they are pulseIn(), delayMicroseconds() ... but limited).

The second one sounds like the only way to make it obvious to the user what they are doing, but still results in unexpected results if they don't know what time unit the API takes for an argument.  The second solution also requires all time values to be explicitly entered which requires more typing than usual.

A third solution could be adding intellisense to the IDE's, helping users know what the argument units are as they are typing it in.  This wouldn't help Sublime Text or other code editor use cases, but the intellisense database could be a common solution for all editors, so a plugin could easily be developed for your editor of choice.

What other solutions are there, and are there other ways of making the first two work more effectively?

m-mcgowan, 2017-02-07:
The user doesn't have to know what time unit the API takes - that's the intent behind making user specify the unit of time, rather than a scalar with no units.  We'd keep the old-style `int` overloads too, but encourage people to use the type-safe api. 

```
class Duration {
private:
   uint32_t quantity;
   
   enum Timescale {
      YEAR, DAY, HOUR, MINUTE, SECOND, MILLISECOND, MICROSECOND, NANOSECOND
   };

   Duration(uint32_t quantity, Timescale scale) {}

public:
   static Duration milliseconds(uint32_t ms) {  return Duration(ms, MILLISECOND); }
   static Duration seconds(uint32_t s) { return Duration(s, SECOND); }
   // ..etc

}

class System {

    int sleep(Duration d, ...) {
           return sleep(d.seconds(), ...);
    }

    int sleep(uint32_t seconds, ...);
}

```

So usage would be:

```
System.sleep(Duration.minutes(5), ...);
```

That's the essence. We might be able to come up with a more pleasant syntax using the new literal support in C++14, or by specifying the quantity and unit separately, e.g.

```
System.sleep(5, MINUTES, ...);
```

monkbroc, 2017-02-07:
I made a proof of concept a while ago of using the C++14 API for type-safe times: `delay(1.5s)` or `System.sleep(SLEEP_MODE_DEEP, 1h);`

https://github.com/spark/firmware/pull/872

avtolstoy, 2020-02-29:
C++14 chrono literals added in #1709

closed 2020-02-29 by avtolstoy