particle-iot/device-os #1119: Core can function as a HID device

AndrewKirkovski, 2016-09-16:
Photon & Electron have HID support, but Core has not. That kinda sucks, Core HAD working HID support in ancient branches.
See: https://github.com/spark/firmware/issues/528
See: http://community.particle.io/t/having-trouble-finding-documentation-for-emulating-a-keyboard/22983/22?u=ryotsuke

technobly, 2017-02-07:
The Core is already very limited in flash and ram space, so adding more features will likely bankrupt the Core's usability.  There is a possibility that this could be implemented as a command line option, but is a low priority.

Ryotsuke, 2017-02-07:
> The Core is already very limited in flash and ram space, so adding more features will likely bankrupt the Core's usability. There is a possibility that this could be implemented as a command line option, but is a low priority.

That is a weird reason. I already have a core running as HID device on older firmware version. But it needs manual changes and is not supported from Web IDE this way.

closed 2020-02-29 by avtolstoy

asquelt, 2021-05-31:
can you point me to ancient firmware with HID enabled? thank you.