particle-iot/device-os #1121: Serial hex print does not print leading zero

dmiddlecamp, 2016-09-20:
```
Serial.print(buffer[i], HEX);
```

when using the HEX print on serial, the leading zero is not printed for single digit values, this is really confusing, and doesn't seem to jive with hex output?

BracWebb, 2016-09-20:
Problem is, how many leading zeros? Always enough to be two digits per length (e.g. 0x0A for 1 byte and 0x0ABC for two bytes?) .. or should the user specify length (e.g. `Serial.print(variable, HEX, 2)`)?
This should do what you need:
`Serial.print(String::format("0x%02X", variable));`
Check out `printf` formating. There is a lot you can make it do!

dmiddlecamp, 2016-09-20:
I think if I'm printing a single character then one leading zero for HEX output makes sense?

dmiddlecamp, 2016-09-20:
I think making the String(foo) wrapper give a nice default makes sense, and then letting users use printf for anything more fancy is the most friendly / intuitive solution?

BracWebb, 2016-09-20:
So if you print a 16-bit variable do you want to make sure it is padded to 4 hex digits and if a 32-bit value to 8 hex digits and so on?

technobly, 2016-09-20:
I think formatting is really a user driven thing.  If you want padding, you must specify it.  If I didn't want padding and the firmware defaults to using it, how would I disable it?

printf/printlnf is also natively supported with Serial now to make this easy.
```
Serial.printlnf("My padded hex byte is 0x%02X", hex_byte);
```

dmiddlecamp, 2016-09-20:
Hi @BracWebb,

  That doesn't sound unreasonable, padding to the data type size makes sense, and would help align padding, etc.  There aren't a huge number of permutations here, right, just a handful?

Thanks,
David

dmiddlecamp, 2016-09-20:
Hi @technobly ,

  I think my concern here is that it isn't intuitive.  I just spent a few hours trying to figure out why my output was garbled, and bytes weren't parsing as expected, only to discover a huge number of leading zeros were missing.  Maybe a good question would be: what does Arduino do for the String / HEX output?

BracWebb, 2016-09-20:
Ha, didn't even know it was natively supported in Serial -- this convo helped me learn something! :)

technobly, 2016-09-20:
@dmiddlecamp I think this is just a consequence of learning C/C++ and we try to stay in line with Arduino/Wiring as much as possible, but in other cases we diverge from it on purpose.  I would bet Arduino doesn't print leading zeros by default based on the data type.  I think an appropriate API would be to add a function that enables leading zeros by default if we wanted to make the firmware add them (beyond what format specifiers already do).  We have a link to format specifiers in one spot in the Docs that I know of (because I added it) http://www.cplusplus.com/reference/cstdio/printf/  but maybe we need more?

m-mcgowan, 2016-09-20:
I would like to verify what arduino does here...and I'm hoping they pad...imho I agree with David about padding according to the datatype, since that is the element of least surprise.

dmiddlecamp, 2016-09-20:
I don't mean to bikeshed here, this just felt inconsistent with our typical "friendly" design, and it cost me a few hours of time.  We have plenty of mechanisms to control padding the typical ways, I just think the default way should be as intuitive as possible.

Checking on Arduino it looks like they have an open issue related to this, so maybe our current Serial.print is compatible with theirs, but maybe we should both change that?

https://github.com/arduino/Arduino/issues/1097

m-mcgowan, 2016-09-20:
As a first resolution to the issue, we can add this info to the docs so it's no longer surprising. And provide the alternative example using printf() that does the padding.

technobly, 2016-09-20:
@dmiddlecamp I just realized your original request was for the HEX argument in Serial.print(variable, HEX); ... and that does make sense.  It's very Arduino/Wiring already by using that API.  I was only pushing back on making any changes to the way C format specifiers work by default, so don't make "%x" add any leading zeros.

Arduino has no docs saying it will add leading zeros https://www.arduino.cc/en/Serial/Print

But it's something we could diverge on :)

dmiddlecamp, 2016-09-20:
Really great conversation overall, thanks everybody for your input and help! :) 

Yeah, I totally agree, changing behavior of something standard like printf would be really silly to do :)

closed 2020-02-29 by avtolstoy