particle-iot/device-os #1124: sscanf doesn't work

dmiddlecamp, 2016-09-23:
pulling my hair out trying to make this work, example app:

```
void setup() {
    Serial.begin(9600);
}

void loop() {
    const char *someString = "3.141592653,1.2345678,72.7751197";

    float latitude, longitude, altitude;
    sscanf(someString, "%f,%f,%f", &latitude, &longitude, &altitude);
    
    Serial.println("running sscanf on string " + String(someString));
    Serial.println(String::format("Parsed into %f, %f, %f ", latitude, longitude, altitude));
    
    delay(5000);
}
```

dmiddlecamp, 2016-09-23:
Expected response:
```
running sscanf on string 3.141592653,1.2345678,72.7751197
Parsed into 3.141592653, 1.2345678, 72.7751197
```

Actual Response:
```
running sscanf on string 3.141592653,1.2345678,72.7751197
Parsed into 0.000000, 0.000000, 0.000000 
```

dmiddlecamp, 2016-09-23:
I've confirmed other people have had this issue while searching the community: https://community.particle.io/t/using-a-particle-function-to-pass-multiple-floats-in-one-function-call/25528

technobly, 2016-09-23:
Yes this is a known issue.  The typical work around can be found here:
https://gist.github.com/technobly/cea340d13cb7848196b5#file-cell-locate-cpp-L121-L122

dmiddlecamp, 2016-09-23:
Hmm, okay, so that workaround translated to my example would be:
```
void setup() {
    Serial.begin(9600);
}

void loop() {
    const char *someString = "3.141592653,1.2345678,72.7751197";

    char one[32], two[32], three[32];
    float latitude, longitude, altitude;
    //sscanf(someString, "%f,%f,%f", &latitude, &longitude, &altitude);
    sscanf(someString, "%[^,],%[^,],%[^,]", one, two, three);
    
    latitude = String(one).toFloat();
    longitude = String(two).toFloat();
    altitude = String(three).toFloat();
    
    Serial.println("running sscanf on string " + String(someString));
    Serial.println(String::format("Tokenized into %s, %s, %s ", one, two, three));
    Serial.println(String::format("Parsed into %f, %f, %f ", latitude, longitude, altitude));
    
    delay(5000);
}
```

technobly, 2016-09-23:
FYI: I tried [changing this briefly](https://github.com/spark/firmware/blob/develop/hal/src/electron/include.mk#L43) to `y` without success or compilation size change... I think the workaround is pretty good though considering we are trying to keep the system firmware lean and mean :)  

- [ ] If we close this without a fix to sscanf, we should document the workaround here: https://docs.particle.io/reference/firmware/photon/#language-syntax

technobly, 2016-09-23:
The above doesn't work with `-u _scanf_float` either, likely because it's only the HAL.  I added this in wiring here and it works: https://github.com/spark/firmware/blob/develop/wiring/inc/include.mk
```
# without LDFLAGS += -u _scanf_float
# 111036	   2568	   5540	 119144	  1d168	../../../build/target/system-part2/platform-10-m/system-part2.elf
# with    LDFLAGS += -u _scanf_float
# 117692	   2568	   5540	 125800	  1eb68	../../../build/target/system-part2/platform-10-m/system-part2.elf
LDFLAGS += -u _scanf_float
```

This adds about 6.5KB of flash though for Electron system part 2 (didn't try Core/Photon)... so that's the tradeoff.

dmiddlecamp, 2016-09-23:
Hmm, that's good to know.  If it's in the system-parts, then I'm guessing we can't easily compile it into user-space on demand.  6.5k seems crazy for parsing floats