particle-iot/device-os #1128: Local compile error "undefined reference to `CallConstructors'"

simoher, 2016-09-25:
< Please summarize your issue here.  Check [x] as many boxes below as you can before submitting your issue.  Thanks -- you rock! >

I got a linker error when trying to compile locally with this make command:
$make all -C C:\Particle\firmware TARGET_DIR= . PLATFORM=photon 

I have the  newest version of the "latest" branch. 

```
c:/particle/toolchain/gcc-arm/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe:linker_stm32f2xx_common.ld:114: warning: memory region `BACKUPSRAM' not declared
../build/target/bootloader/platform-6-lto/startup/startup_stm32f2xx.o: In function `LoopFillZerobss':
c:\Particle\firmware\bootloader/../build/arm/startup/spark_init.S:6: undefined reference to `CallConstructors'
collect2.exe: error: ld returned 1 exit status
```

---

Completeness:
- [X ] Minimum test case added
- [X ] Device, system and user firmware versions stated
- [ ] Particle confirmed

m-mcgowan, 2016-09-25:
Hi @simoher 

This doesn't seem quite right.  :-)  What are you trying to achieve here? 
Please tell us where your application code is stored.

simoher, 2016-09-26:
Hey, I am trying to compile a project outside the firmware folder. 
I am collaborating with a group on this project, and I find it uncomfortable to have to share the whole firmware. Is there a way so that I can avoid, having the project placed in the firmware folder?
The error is also represented when using Chuank's offline compiler for Atom https://github.com/chuank/particle-offline-compiler

technobly, 2016-09-26:
@simoher the simplest way is something like this...
```
cd firmware/modules
make clean all APPDIR=~/code/fw-apps/my_app/ PLATFORM=photon -s COMPILE_LTO=n program-dfu
```
your firmware files would be outside of the firmware repo, and in a folder called `~/code/fw-apps/my_app/`

When Libraries V2 is released (soon), you'll be able to separate your files better... but for now if your project is simply a flat organization of .cpp, .c and .h files this will work.

simoher, 2016-09-27:
Thank you for the reply! Missed the modules part.. Yeah V2 is what I want, have been waiting :) Any chance to get in early on that? I have only been able to find [suda's gist](https://gist.github.com/suda/2c07441a39e04938e003) on it.  Visual Studio fan-boy here... Maybe you know of a good way to use PP with VS? Right now I include the headers from AnuragVasanwala's [visual studio plugin](https://www.hackster.io/AnuragVasanwala/particle-photon-visual-studio-2015-ea22e0) for intelliSense, and the build locally. 

What does the COMPILE_LTO=n flag do? 

I want the build files to get placed in project folder. But when I'm using TARGET_DIR=project_folder I get these errors.

`arm-none-eabi-size --format=berkeley C:\projekts\build/project.PhotonMakefile.elf
   text    data     bss     dec     hex filename
   4924       8    1244    6176    1820 C:\projekts\build/project.PhotonMakefile.elf
arm-none-eabi-objdump -h -S C:\projekts\build/project.PhotonMakefile.elf >C:\projekts\build/project.PhotonMakefile.lst
/usr/bin/sh: C:\projekts\build/project.PhotonMakefile.lst: No such file or directory`

monkbroc, 2016-09-27:
We have a beta for the new project structure (this is part of our overhaul of libraries). I added you to the beta group at http://community.particle.io/ Look for the IDE 2.0 category for instructions on installing the beta release of the CLI and organizing your project according to the new structure. Let us know on the forum how it works for you.

technobly, 2016-09-28:
>Maybe you know of a good way to use PP with VS? Right now I include the headers from AnuragVasanwala's visual studio plugin for intelliSense, and the build locally. Visual Studio fan-boy here...
Perhaps one of these topics on the community can help you? https://community.particle.io/search?q=visual%20studio

>What does the COMPILE_LTO=n flag do?
This disables the Link Time Optimization which produces a slightly larger binary, but one that does not heavily optimize the code.  This makes it easier to debug.  This is the default way Photon, P1 and Electron firmware is compiled.  The Core has COMPILE_LTO=y defined mainly because it really needs all of the extra space it can get.

Looks like you are having issues on Windows.  Windows can be a pain sometimes, and it helps if you convert your PATH variables to DOS 8.3 format.  Use the `dir /x` command to show directories in 8.3 format.  You have to walk up the path and keep using the `dir /x` command, but it doesn't take toooo long.  That may help you.  FYI, your use case works perfectly on the Mac for me, so it should work on windows as well if you can get the paths to play nicely.

simoher, 2016-09-28:
Thanks for adding me! It's not clear to me, whether the particle compile command builds locally or in the cloud. 

Yes, that was the problem. Not sure about the `dir /x` command, to me it returns non-DOS 8.3.
The thing is I'm using the $(ProjectDir) var from VS.. well, converting it is.. at least it works! Great!

ScruffR, 2016-10-01:
`particle compile` does build in the cloud.
It's output reveals that it _uploads_ the files and _downloads_ the binary.

technobly, 2017-02-06:
@simoher would you please sum up what your fix was and close this issue if resolved?  Thanks!

closed 2020-02-29 by avtolstoy