particle-iot/device-os #1129: Custom Logger Category: Compile Error (multiple definition)

LukeUSMC, 2016-09-27:
Photon
System 6.0rc2
Logger Feature
User Firmware
```cpp
// This #include statement was automatically added by the Particle IDE.
#include "Custom.h"

#include "application.h"

#define LOGGING_ON TRUE

int log_tempo = 30000;
int next_log;

Custom myObj;

void setup() {
  if (LOGGING_ON){
    Serial.begin(115200);
    SerialLogHandler logHandler(LOG_LEVEL_WARN, {
        { "app", LOG_LEVEL_TRACE },
        { "app.custom", LOG_LEVEL_TRACE }
    });
    Log.trace("Logging Enabled, moving to Loop");
  }
}

void loop() {
  if (millis() > next_log){
    Log.info("Logged in loop.");
    myObj.some_function();
    next_log = millis() + log_tempo;
    }
}
```
Custom.h
```cpp
#include "application.h"

#ifndef Custom_h
#define Custom_h

Logger myLog("app.custom");

class Custom {
private:
    
public:
    Custom();
    ~Custom();
    
    void some_function();
protected:

};

#endif /* Custom_h */
```
Custom.cpp
```cpp

#include "Custom.h"

Custom::Custom() {

}

Custom::~Custom() {

}

void Custom::some_function(){
  myLog.info("Log from custom class.");
}
```
Exact error:
```
In function `spark::StreamLogHandler::write(char const*, unsigned int)':
/firmware/user/../wiring/inc/spark_wiring_logging.h:475: multiple definition of `myLog'
../../../build/target/user/platform-6-m/workspace//libuser.a(Custom.o):/firmware/user/../wiring/inc/spark_wiring_logging.h:475: first defined here
collect2: error: ld returned 1 exit status
../../../build/module.mk:216: recipe for target '/workspace/target/workspace.elf' failed
make[1]: Leaving directory '/firmware/modules/photon/user-part'
make[1]: *** [/workspace/target/workspace.elf] Error 1
../build/recurse.mk:11: recipe for target 'modules/photon/user-part' failed
make: *** [modules/photon/user-part] Error 2
```
---

Completeness:
- [ X] Minimum test case added
- [ X] Device, system and user firmware versions stated
- [ ] Particle confirmed

monkbroc, 2016-09-27:
Hi Luke,

I know that the logHandler is meant to be at the top level, not inside a function like `setup()` because in that case the object will get destroyed when `setup()` returns.

```
#if LOGGING_ON
SerialLogHandler logHandler(LOG_LEVEL_WARN, {
    { "app", LOG_LEVEL_TRACE },
    { "app.custom", LOG_LEVEL_TRACE }
});
#endif

void setup() {
}
```

For the multiple defines, you need to move `Logger myLog("app.custom");` to the CPP source file instead of the header file. You can also use the C `static` keyword to tell the compiler that `myLog` should only be visible within its own file. This way you can use a different `myLog` variable in another  CPP file without getting a duplicate variable error.

LukeUSMC, 2016-09-28:
Thank you...I have it working now.  Just an FYI and barely worth mentioning but any use of logSomething in Build the log portion of the var is highlighted (I believe indicating the log function in c++) but the remainder of the declaration is not highlighted.  For example:
"logHandler(LOG_LEVEL_WARN," will display with log in blue and Handler in white, I am not sure if this would lead to confusion for some or not.  Thanks again.

closed 2016-09-28 by LukeUSMC