particle-iot/device-os #1131: SoftAP read function issue.

Suprazz, 2016-10-01:
Two continuous body->read calls in page handler of the SoftAP application won't return consecutive data except if the read is 1152 bytes long.

Example: This work perfectly:
uint8_t contentData[1152];
uint8_t contentData2[1152];
int datalength = body->read(contentData, sizeof(contentData));
int datalength2 = body->read(contentData2, sizeof(contentData));

With this code, there is data lost between the 2 buffers.
uint8_t contentData[512];
uint8_t contentData2[512];
int datalength = body->read(contentData, sizeof(contentData));
int datalength2 = body->read(contentData2, sizeof(contentData));

Refer to this thread for more info: https://community.particle.io/t/firmware-upgrade-via-softap/26198/11
---

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

m-mcgowan, 2016-10-01:
Thank you for the  issue report. We'll investigate and try to determine the cause of the issue, and in the meantime we will update the documentation to reflect this use case.

bkobkobko, 2016-10-01:
The Community thread is here:

http://community.particle.io/t/firmware-upgrade-via-softap/26198

The work-around is obivously to use a 1152-byte (TCP_MSS) sized buffer.

The inline read function looks right to me so I think it is something in the WICED source that you can't see without an NDA.

m-mcgowan, 2016-10-01:
Thanks bko. The WICED http server is implemented by using TCP as packets rather than as a stream (which is performant, but the wrong abstraction imho.) So it doesn't surprise me these kind of things crop up as a result. 

I'm not sure we can fix it without rewriting the http server, but we can at least document it and hope WICED have a fix in a future release. We are expecting a new drop from them soon.

avtolstoy, 2020-03-03:
This should be resolved as of 0.7.0+.

closed 2020-03-03 by avtolstoy