particle-iot/device-os #1136: [Core 0.5.3ff]: Interrupts are disabled by default

ScruffR, 2016-10-08:
As discussed here, 0.5.3 seems to have introduced an unexpected behaviour with interrupts on Cores
https://community.particle.io/t/interrupts-disabled-on-core-fw-0-5-3-and-later/26393

Only after adding `interrupts();` they start firing.

```cpp
void isr()
{
  digitalWrite(D7, !digitalRead(D7));
}

void setup() {
  pinMode(D7, OUTPUT);
  pinMode(D3, INPUT_PULLDOWN);
  attachInterrupt(D3, isr, CHANGE);
  //attachInterrupt(D3, isr, RISING);
  //attachInterrupt(D3, isr, FALLING);
  //interrupts();
}

void loop() { }
```

---

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

gmazzo, 2016-11-01:
I've spent three days thinking I was doing something wrong. In deed, add `interrupts();` did the trick. 

The docs says it's enabled by default, so it has to be an issue:
https://docs.particle.io/reference/firmware/core/#interrupts

ScruffR, 2016-11-01:
Thanks for confirming 👍

ScruffR, 2017-01-05:
It's a while now and still no Particle confirmation?
An new case where user got bitten by this still unaddressed issue
 https://community.particle.io/t/unable-to-get-results-from-dht22/28361

But there would be a simple cure (I guess)

technobly, 2017-02-06:
Thanks for the great test case @ScruffR !

I can confirm this is working as expected on 0.5.2, and broken on 0.5.3 (requiring `interrupts();` for the ISR to fire).

PR #1255 (2017-05-04): Fix for attachInterrupt() on the Core
This PR fixes #1136 by reverting what I believe was an accidental change introduced by this PR: https://github.com/spark/firmware/pull/1076.

I don't have a Core device personally, but I've managed to reproduce symptoms of the issue on the Photon, so any help with testing is appreciated.

---

Doneness:

- [X] Contributor has signed CLA
- [X] Problem and Solution clearly stated
- [ ] Code peer reviewed
- [ ] API tests compiled
- [ ] Run unit/integration/application tests on device
- [ ] Add documentation
- [ ] Add to CHANGELOG.md after merging (add links t [... cut]
closed 2017-05-04 by technobly