particle-iot/device-os #1137: udpClient.sendPacket starts taking 2000 ms and then the Photon freezes

jnm2, 2016-10-10:
We have an issue that I don't know how to get started diagnosing. Photon will lose WiFi and then freeze over the course of a few days. I added logging and found that it always freezes inside `udpClient.sendPacket`. (`udpClient` is a global `UDP` variable.)

I wondered if a buffer was overfull, so I made the code detect when WiFi is lost (!WiFi.ready()` and call `udpClient.Stop()` and not send any more packets until WiFi is ready again, upon which it calls `udpClient.begin(port)` and resumes sending packets.

What should I do to diagnose further?

Firmware version 0.5.3.

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

jnm2, 2016-10-11:
I removed `udpClient.stop()` and `udpClient.begin(port)` and it seems to be working. Are there likely to be other remaining issues?

The documentation on `begin` and `stop` is very sparse and if they should not be used in this scenario, it would be good to know in exactly which scenarios they should be called.

jnm2, 2016-10-11:
I removed `udpClient.stop()` and `udpClient.begin(port)` and now `sendPacket` doesn't slow down and freeze, but it also stops actually sending any data.

I need some suggestions. Is there a buffer that needs to be cleared?

tlangmo, 2016-10-12:
I encountered a lot of problems with reconnection as well, including yours. I only can give some more insight, but no real solution. It also does not appear to happen all the time.

The 2000 ms delay is hardcoded in the WICED sdk in` internal_packet_create` It tries 2000 ms to allocate a packet to send, after that is reports a WICED_TIMEOUT. I don't know the internals of how all things work together but I think it is ok that it happens. The timeout can be caused by a wifi glitch. 

After the particle firmware detects that the WiFi is really unreachable via a Watchdog timer, it then reconnects the WiFi. In that process, all existing sockets get destroyed in `socket_hal.cpp:socket_close_all()`. 

Long story short, you definitely need to re-create your resources as well, as you have seen yourself already.

Can you try to narrow down the problem and provide a test script? It's important to now the whole config, especially with System threading. In my office I use an Access Point which I can quickly turn on and off to test faster. 

Btw, I am using the most current develop branch, so I do not think the problem was already fixed since 0.5.3

technobly, 2017-02-06:
@jnm2 could you please provide a minimum code example and describe the observed behavior vs. the expected behavior with the example you provide?  From your description is sounds like we could recreate more or less, but it would be extremely helpful to have a solid example.

jnm2, 2017-02-19:
@technobly I no longer have access to a Photon, since they were not owned by me and the project is finished. :-( Other than that, I would be happy to provide a minimum repro. Particularly happy if you provided me with a Photon :-)

jnm2, 2019-02-07:
It this was tracking work that you still want to do, please feel free to reopen! Otherwise, cleaning up due to inactivity.