particle-iot/device-os #1141: WICED TCP  limit of 65535 bytes in one call.

Suprazz, 2016-10-17:
The WICED TCP reader eventually uses an uint16 as the size, so there is a fundamental limit of 65535 bytes in one call.

Reference: https://community.particle.io/t/firmware-upgrade-via-softap/26198/18

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

technobly, 2017-02-06:
I'm not sure we can or should fix this.  I would suspect the 65536 bytes are stored in SRAM.  Photon/P1/Electron only have 128KB total and all of that is not even available to the user by default since some is used by the system.

Could you provide a bare minimum example of how this should work for you.  I'm wondering if you are also allocating a buffer of your own in SRAM.

Suprazz, 2017-02-06:
simply increase the uint16 to unint32.
This doesnt mean that the 65k is load into ram. The data is partially read until we can reach 65k and after that everything screw up. Because is a tcp socket, the tcp protocol can stop to receive data until there is free space in the buffer.

m-mcgowan, 2017-02-06:
Hi @Suprazz. I hear what you're saying, and while the TCP might at present be the limiting factor, I feel there's a lot of context missing here (and the forum post rather long!).  Could you explain the high-level goal of what you're trying to achieve, and how you're presently doing it, and what bug is being encountered. 

Even with the 64k limit on TCP, there are (presumably) many layers between what you're accessing and the underlying TCP stack, so plenty of opportunity for us to split the data into smaller chunks on route.

Suprazz, 2017-02-06:
I'm doing a firmware upgrade via the SoftAP http webpages. I do a form submit and the firmware is send like this. But because the firmware can be bigger than 65k, I have to use javascript functions to split the upload in multiple calls of 60k (and keeping 5k for headers just in case).

technobly, 2017-02-07:
It sounds like this use case should be investigated since it should be a fairly common case.  If you have time to test that simple change of uint16 to unint32, please let us know if it solves the issue for you.

Suprazz, 2017-02-07:
I dont know how to compile system part 2 with debug options(serial debugging) so I can really know that the flash worked and that the code is running my version of system part 2 to debug and confirm that it's working.

bkobkobko, 2017-02-07:
It's WICED issue so I don't see how you can just compile the system firmware and fix this. Maybe a wrapper layer could be written as Mat said above, but it is not just a quick change and compile.

firmware/hal/src/photon/libraries/daemons/HTTP_server/http_server.h

It looks like the WICED server does support chunked transfers, so I would recommend investigating that also.

Suprazz, 2017-02-07:
Thanks. Yes, there is also an other issue where it's not possible to do 2 consecutive read without loosing data if the read size is different that 1152 bytes.

closed 2020-03-03 by avtolstoy