particle-iot/device-os #1143: Unexpected difference in behaviour when passing unsigned long instead of long

ScruffR, 2016-10-19:
See this discussion
https://community.particle.io/t/electron-deep-sleep-and-never-wake-up/26613

Apparently `System.sleep(SLEEP_MODE_DEEP, wakeTime)` behaves differently whether `wakeTime` is declared as `unsigned long` vs. `long` which doesn't seem to make sense as long the passed value doesn't leave the positive range of long.

How come? 
Is this unexpected distinction limited to `System.sleep()` or will the same root cause hit on other ends too?

---

Completeness:
- [x] Minimum test case added (see linked thread)
- [x] Device, system and user firmware versions stated (see linked thread)
- [ ] Particle confirmed

technobly, 2017-02-06:
I don't seem to be able to reproduce this issue with the following test app.  `tone()` so I don't have to stare at it for verification that it's working.

```c
SYSTEM_MODE(SEMI_AUTOMATIC);
STARTUP(System.enableFeature(FEATURE_RETAINED_MEMORY);)

retained int state = 0; // 0 = deep sleep unsigned long, 1 = deep sleep long

void setup()
{
    pinMode(D0, OUTPUT);
    if (++state > 4) state = 1;

    if (state == 1) {
        tone(D0, 500, 1000);
        delay(1000);
        System.sleep(SLEEP_MODE_DEEP, 10L);
    }
    else if (state == 2) {
        tone(D0, 1000, 1000);
        delay(1000);
        System.sleep(SLEEP_MODE_DEEP, 10UL);
    }
    else if (state == 3) {
        tone(D0, 1500, 1000);
        delay(1000);
        unsigned long sleep_time = 10;
        System.sleep(SLEEP_MODE_DEEP, sleep_time);
    }
    else if (state == 4) {
        tone(D0, 2000, 1000);
        delay(1000);
        long sleep_time = 10;
        System.sleep(SLEEP_MODE_DEEP, sleep_time);
    }
    while(1);
}
```

avtolstoy, 2020-03-03:
This issue is resolved in current releases.

closed 2020-03-03 by avtolstoy