particle-iot/device-os #1149: Sleep API Refactoring

technobly, 2016-10-27:
## **Goal:** 
>refactor Sleep API's to remove problems, make intuitive and easy to remember.

**This issue is related to these previous issues:**
#501 - Additional sleep modes and sleep API review
#502 - Wiring API supports multiple Network types 
#251 - Comment on Network.on/off vs WiFi.on/off

**As a recap, the current Sleep API looks like this:**

| API | What it does | Network | Application |
|:-:|:-:|:-:|:-:|
| System.sleep(60) | Put the NETWORK module to sleep for 60s. Application code continues running. | OFF | RUNNING |
| System.sleep(D2, RISING, 60) | Pause execution, using less power. Resume execution with the next statement after 60s, or when D2 transitions from LOW to HIGH, whichever is sooner. | OFF | PAUSED |
| System.sleep(D2, RISING, 60, SLEEP_NETWORK_STANDBY) | Same as above with network connection still active.  Only available on Electron. | ON | PAUSED |
| System.sleep(SLEEP_MODE_DEEP, 60) | Stop execution, using much less power. Restart the system after 60 seconds or on a rising edge on the WKP pin (photon/P1/electron only) or falling edge on RESET pin.  The application restarts - memory is cleared, setup() is called.) | OFF | TERMINATED |
| System.sleep(SLEEP_MODE_DEEP, 60, SLEEP_NETWORK_STANDBY) | Same as above with network connection still active.  Only available on Electron. | ON | TERMINATED |
| System.sleep( SLEEP_MODE_SOFTPOWERDOWN, 60) | Same as above with network connection off and also sleeps the Fuel Gauge further reducing power.  Only available on Electron. | OFF | TERMINATED |

**The current Sleep API has these issues:**

| API | Issue |
|:-:|:-:|
| System.sleep(60) | Doesn’t stop application code.  Not intuitive. |
| System.sleep(D2… | Doesn’t intuitively convey STOP mode which is a STM naming convention. This corresponds to application level naming convention PAUSE. |
| System.sleep(SLEEP_MODE_DEEP... | Doesn’t intuitively convey STANDBY mode which is a STM naming convention. Hard to remember. This corresponds to application level naming convention DEEPSLEEP. |
| System.sleep( SLEEP_MODE_SOFTPOWERDOWN... | Doesn’t intuitively convey STANDBY mode which is a STM naming convention. Hard to remember. This corresponds to application level naming convention DEEPSLEEP.  In addition to sleeping the Network and MCU, this also sleeps the Fuel Gauge further reducing power.  Electron only. |

**Proposed APIs:**

| API | What it does |
|:-:|:-:|
| Network.on() | **new:** Turns WiFi/Cellular/Ethernet on.  WiFi.on(), etc.. still works. |
| Network.off() | **new:** Turns WiFi/Cellular/Ethernet off. WiFi.off(), etc.. still works. |
| Network.sleep(60) | **refactor:** Turns off the network connection for 60 seconds.  Application code continues running. System.sleep(60) would be deprecated in time. |
| System.pause(60) | **new:** Halts the system code in place for 60 seconds.  No interrupt set to wake up the system prematurely. |
| System.pause(D2, RISING, 60) | **refactor:** Halts the system code in place, network device turned off by default |
| System.pause(D2, RISING, 60, SLEEP_NETWORK_ON) | **refactor:** Halts the system code in place, network device left on. |
| System.deepsleep(60) | **refactor:** Lowest power state, network device turned off by default. Fuel Gauge on electron sleep by default. |
| System.deepsleep(60, SLEEP_NETWORK_ON) | **refactor:** Lowest power state, network device turned off by default. |
| System.deepsleep(60, SLEEP_FUELGAUGE_ON) | **refactor:** Lowest power state, network device turned off by default. Fuel Gauge on electron remains active monitor battery. |
Use SLEEP_NETWORK_ON as the new default for keeping the network on when sleeping the MCU | **refactor:** Keep SLEEP_NETWORK_STANDBY available to do the same thing, in the future SLEEP_NETWORK_STANDBY may imply a different function for a different device |
| Existing APIs associated with above | "Depreciated" Warning (to be removed in v1.0.0) |

**Part of this effort will include use case examples and summary tables in [Docs](https://docs.particle.io/reference/firmware/electron/#sleep-sleep-)**
- [ ] Link to Docs issue (when created): _add_link_

---

Completeness:
- [x] Minimum test case added
- [x] Device, system and user firmware versions stated
- [x] Particle confirmed

rickkas7, 2016-10-27:
Much improved. I like it! 
One minor thing: should it be `System.deepSleep()` with camel case? I can see reasons to make it either way, but I think I prefer mixed-case.

technobly, 2016-10-27:
I was hoping deepsleep was considered as a state, and thus a one word item.  We've also all been typing it as one work in SLEEP_MODE_DEEPSLEEP, but if you can justify it more as two words then it should be camelCase() :)  I was also pooling my thoughts that it was one word because of the Wi-Fi Powersave feature is referred to by [WICED as one word](http://cloudconnectkits.org/sites/default/files/Powersave%20App%20Note%20(MMPWICED-AN104-R).pdf) and thus our API for that will be similar like this WiFi.enablePowersave() and WiFi.powersaveEnabled().  Maybe I'm over thinking it though.

I could see an argument for alwaysUseCamelCase becauseWeAlwaysUseCamelCase as well and I could get behind that (and we should scrub the entire API of outliers then) :)

technobly, 2016-10-28:
I updated System.deepsleep() to System.deepSleep() and we'll try to keep consistency with camelCase API in firmware from now on ;-)  That way if you remember the name, you will remember exactly how to type it.  This is part of the original GOAL as well!  Thanks for the suggestion @rickkas7

technobly, 2016-10-28:
After some more discussion I updated the original post by changing `System.pause()` to `System.pauseSleep()`.  Pause by itself did not imply sleep of the system/network as well, and sleep by itself does not imply pausing of code execution.

So the two main sleep modes would be `System.pauseSleep()` and `System.deepSleep()`.  They are both searchable in Docs with the word "sleep".  They both imply system sleeping, and by default the network device would be put to sleep as well.  One now conveys it’s pausing to sleep and the other hopefully conveys deep full sleep which when awoken from is a brand new day ☀️

ScruffR, 2016-10-29:
For "consistence" when there is a non-interrupt version (e.g. `System.pause(60)`) there should also be a non-timed `System.pause(D0)` which only wakes on interrupt.
Also a way to have multiple interrupt sources set had come up in the community - the workaround to OR the sources into one dedicated pin would require sacrificing one GPIO otherwise useful differently.
This could be either done by passing a pin array (requiring all to share the same edge), two arrays for pin and edge, a pin/edge struct array, or by having a new `System.addWakeSource(pin, edge [, optionalPullResistor])` with a subsequent `System.pause(uint32_t seconds=0)` (0 for only wake on interrupt, no timeout).
The latter may also open a way to add other interrupt sources (although can't think of one yet) for waking the device.

mfogel, 2016-10-31:
This is great to see. I just worked through a good portion of the current confusing situation with sleep modes on the electron myself just now. Nice to see it all summarized on this ticket.

One thought/suggestion on the proposed api:

If `Wifi.on()`/`Wifi.off()`/`Cellular.on()`/`Cellular.off()` are not going to be deprecated, then the only thing the new top-level `Network` object really provides me is `Network.sleep()`... right? Might it be simpler for users if instead this functionality is provided as `Wifi.sleep()` and `Cellular.sleep()` and a new top-level object is avoided?

Along those lines, I'm not sure if it's on Particle's roadmap or not, but if there are plans to release a device with multiple ways of connecting to the internet (ie. Cellular and Wifi) then having this one singular `Network` object might be confusing. I guess `Network.sleep()` would then sleep all the network connections? But then you would also want to be able to sleep individual connections as well - so you'd need `Wifi.sleep()`/`Cellular.sleep()` as well or something like them.

avtolstoy, 2020-03-03:
#1986

closed 2020-03-03 by avtolstoy