particle-iot/device-os #1152: Add arduino SPISettings, start/endTransaction() functions to firmware

pkourany, 2016-10-30:
The existing Particle firmware supports the basic Arduino SPI functions.  However, when there are two or more SPI devices on the bus, they tend not to play nice.  In arduino land, this is addressed using SPISettings() object, and accompanying beginTransaction() and endTransaction() functions to set and reset the SPI bus settings before and after each transaction (the SDFat library uses simple versions of these!).  Vendors are writing their libraries using these functions more often (and for good reasons).

An ideal implementation would "push" the active SPI control register (2) values on SPI.beginTransaction() and "pop" them on SPI.endTransaction() to be the ultimate best SPI citizen.

[SPISettings() ](https://www.arduino.cc/en/Reference/SPISettings) is an SPI friend class while [beginTransaction()](https://www.arduino.cc/en/Reference/SPIbeginTransaction) and [endTransaction()](https://www.arduino.cc/en/Reference/SPIendTransaction) are SPI class functions.

I have attached copies of the Arduino (v1.6.14) SPI library files for reference

[Arduino SPI libs.zip](https://github.com/spark/firmware/files/560237/Arduino.SPI.libs.zip)

---

Completeness:
- [ ] Minimum test case added
- [ ] Device, system and user firmware versions stated
- [ ] Particle confirmed

whatnick, 2016-12-13:
Would this be available in the particle online compiler soon ?

pkourany, 2016-12-13:
@whatnick, not until the next release of system firmware.

avtolstoy, 2017-01-20:
Implemented in #1188, included in 0.6.1-rc.1

closed 2017-01-20 by avtolstoy

ScruffR, 2017-09-06:
There  is a regression since 0.6.2-rc.1
https://github.com/spark/firmware/issues/1381