particle-iot/device-os #1163: System Eco Mode

technobly, 2016-11-07:
This is a proposal for a new low power feature called **System Eco mode**.

**Related to discussion here:**
#935 - Low Power Run mode

System Eco mode (enabled) drops the STM32 processor clock speed to _XX_ MHz and disables all peripherals except (_name the peripherals required for simple timing and GPIO_).  Code continues to execute in this mode at a slower pace based on the system clock speed.  Some commands are not functional when the clock speed is this low (_provide a table of available commands_).

System Eco mode (disabled) restores the STM32 processor clock speed to 120 MHz and enables all peripheral support.

System Eco mode can also be enabled for a time period defined in seconds.

A future API will also add individual support for configuring the clock and peripherals individually (see proposed API).  For the first version of this feature, we only wish to enable a minimal use case by default that provides maximum power reduction and minimum GPIO/Timer support for sensor and user input.

### Proposed API:

See issue https://github.com/spark/firmware/issues/1150 for the API.

---

Completeness:
- [x] Minimum test case added
- [x] Device, system and user firmware versions stated
- [x] Particle confirmed

dmiddlecamp, 2018-05-09:
getting more bumps lately for being able to downclock the mcu speed on our various boards :)

dmiddlecamp, 2019-03-06:
Getting bumps for this again for folks developing on e-series running off batteries