particle-iot/device-os #1166: Keep device connected during download of OTA updates

ScruffR, 2016-11-10:
As mentioned as a side not in #1165

I recently noticed several threads wondering why pending updates of the Console won't get delivered properly.
e.g. 
http://community.particle.io/t/product-firmware-cloud-update-timing/27064
http://community.particle.io/t/new-firmware-for-products-gets-never-flashed/26621
http://community.particle.io/t/flashing-new-firmware-to-production-devices/27147

My usual response is, check `System.updatesPending()` (which is not that well documented and obviously needs some time to give you correct results - which seems completely undocumented) and if `true` keep hanging around for about a minute for OTA to download.
But that doesn't seem to be that convenient IMHO.

So I'd think it'd be good to have a started OTA update block any attempts that would interfere with that update.
Namely
```
  System.sleep();
  WiFi.off();
  WiFi.disconnect();
  WiFi.listen(); // ???
  Cellular.off();
  Cellular.disconnect();
  Particle.disconnect();
  // and whatever I forgot
```
It might be thinkable to add a parameter `boolean force = false`  to give the user contol if in his case the call should be executed regardless of OTA updates (or pending publishes as in #1165) or be cooperative.

technobly, 2017-02-06:
Thanks for the note on this!  We have System.updatesPending() on the list of things to document better, and enhance.   I'm leaving this unconfirmed for now since it's unclear what the resolution should be, but there definitely needs to be an enhancement made.

ScruffR, 2017-06-22:
Just to add to the list of victims to this issue
https://community.particle.io/t/photon-ota-vs-system-sleep/33954
And myself for a contract work I recently discussed with Dave

ScruffR, 2018-04-03:
Could this be a way forward?
https://github.com/particle-iot/firmware/issues/1285#issuecomment-376497903