particle-iot/device-os #1168: Please add a way to create low latency interrupts

ScruffR, 2016-11-13:
There are occasions where people would be willing to trade HW independents (HAL) for speed.
One of the occasions are interrupts.
The HAL seems to introduce a 5µs latency of EXTI and HW timer interrupts, which limits the usefulness for some projects considerably.
@pkourany and I had brought this forward before (several times), so I wanted to have this listed here to keep it on the radar.
I tried this, but the bare metal approach didn't work at all and the `attachInterrupt()` exhibited a minimum latency of 3µs where some 100ns should be pissible.
```/**
 *	External interrupts example
 *
 *	@author 	Tilen Majerle
 *	@email		tilen@majerle.eu
 *	@website	http://stm32f4-discovery.com
 *	@ide		Keil uVision 5
 *  https://stm32f4-discovery.net/2014/08/stm32f4-external-interrupts-tutorial/
 */
//#include "stm32f2xx.h"
//#include "stm32f2xx_exti.h"
//#include "stm32f2xx_syscfg.h"
//#include "misc.h"

volatile uint32_t d2;
volatile uint32_t d3;
volatile uint32_t d4;

volatile uint32_t us;
volatile uint32_t us2;
volatile uint32_t us3;
volatile uint32_t us4;

uint32_t offset;

#undef EXTI4_IRQHandler
#undef EXTI9_5_IRQHandler

/* Set interrupt handlers */
/* Handle PB4 interrupt */
void EXTI4_IRQHandler(void) {
	/* Make sure that interrupt flag is set */
	if (EXTI_GetITStatus(EXTI_Line4) != RESET) {
		/* Do your stuff when PB4 is changed */
		
        us3 = SYSTEM_TICK_COUNTER;
        d3++;

		/* Clear interrupt flag */
		EXTI_ClearITPendingBit(EXTI_Line4);
	}
}

/* Handle PB5 interrupt */
void EXTI9_5_IRQHandler(void) {
	us2 = SYSTEM_TICK_COUNTER;
    d2++;

	/* Make sure that interrupt flag is set */
	if (EXTI_GetITStatus(EXTI_Line5) != RESET) {
		/* Do your stuff when PB12 is changed */
		

		/* Clear interrupt flag */
		EXTI_ClearITPendingBit(EXTI_Line5);
	}
}

/* Configure D3 for bare metal interrupt */
void Configure_PB4(void) {
	/* Set variables used */
	GPIO_InitTypeDef GPIO_InitStruct;
	EXTI_InitTypeDef EXTI_InitStruct;
	NVIC_InitTypeDef NVIC_InitStruct;
	
	/* Enable clock for GPIOD */
	RCC_AHB1PeriphClockCmd(RCC_AHB1Periph_GPIOB, ENABLE);
	/* Enable clock for SYSCFG */
	RCC_APB2PeriphClockCmd(RCC_APB2Periph_SYSCFG, ENABLE);
	
	/* Set pin as input */
	GPIO_InitStruct.GPIO_Mode = GPIO_Mode_IN;
	GPIO_InitStruct.GPIO_OType = GPIO_OType_PP;
	GPIO_InitStruct.GPIO_Pin = GPIO_Pin_4;
	//GPIO_InitStruct.GPIO_PuPd = GPIO_PuPd_UP;
	GPIO_InitStruct.GPIO_PuPd = GPIO_PuPd_DOWN;
	GPIO_InitStruct.GPIO_Speed = GPIO_Speed_100MHz;
	GPIO_Init(GPIOB, &GPIO_InitStruct);
	
	/* Tell system that you will use PB4 for EXTI_Line4 */
	SYSCFG_EXTILineConfig(EXTI_PortSourceGPIOB, EXTI_PinSource4);
	
	/* PB4 is connected to EXTI_Line4 */
	EXTI_InitStruct.EXTI_Line = EXTI_Line4;
	/* Enable interrupt */
	EXTI_InitStruct.EXTI_LineCmd = ENABLE;
	/* Interrupt mode */
	EXTI_InitStruct.EXTI_Mode = EXTI_Mode_Interrupt;
	/* Triggers on rising and falling edge */
	EXTI_InitStruct.EXTI_Trigger = EXTI_Trigger_Rising;
	//EXTI_InitStruct.EXTI_Trigger = EXTI_Trigger_Falling;
	//EXTI_InitStruct.EXTI_Trigger = EXTI_Trigger_Rising_Falling;
	/* Add to EXTI */
	EXTI_Init(&EXTI_InitStruct);

	/* Add IRQ vector to NVIC */
	/* PB4is connected to EXTI_Line4, which has EXTI4_IRQn vector */
	NVIC_InitStruct.NVIC_IRQChannel = EXTI4_IRQn;
	/* Set priority */
	NVIC_InitStruct.NVIC_IRQChannelPreemptionPriority = 0x00;
	/* Set sub priority */
	NVIC_InitStruct.NVIC_IRQChannelSubPriority = 0x00;
	/* Enable interrupt */
	NVIC_InitStruct.NVIC_IRQChannelCmd = ENABLE;
	/* Add to NVIC */
	NVIC_Init(&NVIC_InitStruct);
}

void Configure_PB5(void) {
	/* Set variables used */
	GPIO_InitTypeDef GPIO_InitStruct;
	EXTI_InitTypeDef EXTI_InitStruct;
	NVIC_InitTypeDef NVIC_InitStruct;
	
	/* Enable clock for GPIOB */
	RCC_AHB1PeriphClockCmd(RCC_AHB1Periph_GPIOB, ENABLE);
	/* Enable clock for SYSCFG */
	RCC_APB2PeriphClockCmd(RCC_APB2Periph_SYSCFG, ENABLE);
	
	/* Set pin as input */
	GPIO_InitStruct.GPIO_Mode = GPIO_Mode_IN;
	GPIO_InitStruct.GPIO_OType = GPIO_OType_PP;
	GPIO_InitStruct.GPIO_Pin = GPIO_Pin_5;
	//GPIO_InitStruct.GPIO_PuPd = GPIO_PuPd_UP;
	GPIO_InitStruct.GPIO_PuPd = GPIO_PuPd_DOWN;
	GPIO_InitStruct.GPIO_Speed = GPIO_Speed_100MHz;
	GPIO_Init(GPIOB, &GPIO_InitStruct);
	
	/* Tell system that you will use PB5 for EXTI_Line5 */
	SYSCFG_EXTILineConfig(EXTI_PortSourceGPIOB, EXTI_PinSource5);
	
	/* PB12 is connected to EXTI_Line5 */
	EXTI_InitStruct.EXTI_Line = EXTI_Line5;
	/* Enable interrupt */
	EXTI_InitStruct.EXTI_LineCmd = ENABLE;
	/* Interrupt mode */
	EXTI_InitStruct.EXTI_Mode = EXTI_Mode_Interrupt;
	/* Triggers on rising and falling edge */
	EXTI_InitStruct.EXTI_Trigger = EXTI_Trigger_Rising;
	//EXTI_InitStruct.EXTI_Trigger = EXTI_Trigger_Falling;
	//EXTI_InitStruct.EXTI_Trigger = EXTI_Trigger_Rising_Falling;
	/* Add to EXTI */
	EXTI_Init(&EXTI_InitStruct);

	/* Add IRQ vector to NVIC */
	/* PB12 is connected to EXTI_Line5, which has EXTI9_5_IRQn vector */
	NVIC_InitStruct.NVIC_IRQChannel = EXTI9_5_IRQn;
	//NVIC_InitStruct.NVIC_IRQChannel = EXTI15_10_IRQn;
	/* Set priority */
	NVIC_InitStruct.NVIC_IRQChannelPreemptionPriority = 0x00;
	/* Set sub priority */
	NVIC_InitStruct.NVIC_IRQChannelSubPriority = 0x01;
	/* Enable interrupt */
	NVIC_InitStruct.NVIC_IRQChannelCmd = ENABLE;
	/* Add to NVIC */
	NVIC_Init(&NVIC_InitStruct);
}

void intD4()
{
    us4 = SYSTEM_TICK_COUNTER;
    d4++;
    digitalWrite(D7, !digitalRead(D7));
}

int triggerInterrupt(String dmy)
{
    us2=us3=us4=0;
    us=SYSTEM_TICK_COUNTER;
    pinSetFast(D0);
    delayMicroseconds(10);
    pinResetFast(D0);
    
    return d4;
}

void dmy() { }

void setup() 
{
    /* System init */
	//SystemInit(); // BREAKS CLOUD
	/* Configure D3 as interrupt */
	Configure_PB4();
	/* Configure D2 as interrupt */
	Configure_PB5();
	
    Particle.function("trig", triggerInterrupt);
    
    pinMode(D0, OUTPUT);
    pinMode(D7, OUTPUT);

    us=SYSTEM_TICK_COUNTER;
    pinResetFast(D0);
    offset = SYSTEM_TICK_COUNTER - us;
    
    pinMode(D4, INPUT_PULLDOWN);
    attachInterrupt(D4, intD4, RISING);
}

void loop()
{
/*
    if(us4)
    {
        char txt[256];
        snprintf(txt, sizeof(txt), "%12d ns (%5d): %12d ns (%5d): %12d ns (%5d)"
                                 , (us2-us-offset)*1000/SYSTEM_US_TICKS, d2
                                 , (us3-us-offset)*1000/SYSTEM_US_TICKS, d3
                                 , (us4-us-offset)*1000/SYSTEM_US_TICKS, d4
                );
        Serial.println(txt);
        Particle.publish("intTimes", txt, PRIVATE);
        us4 = 0;
    }
*/
  //Particle.publish("test", String(millis()), PRIVATE);
  //delay(2000);
}
```
_(tested on Photon 0.6.0-rc.2)

---

Completeness:
- [X] Minimum test case added
- [X] Device, system and user firmware versions stated
- [ ] Particle confirmed

avtolstoy, 2016-11-13:
@ScruffR 
> Currently the NVIC table seems to reside in flash, preventing any updates.
> How can we relocate it back to RAM without breaking HAL and why does it have to be in flash to start with?

I don't think that NVIC Vector table resides in flash (at least not on Photon/P1/Electron). You can get its address by accessing `SCB->VTOR` register and you should be able to override interrupt handlers there directly.

ScruffR, 2016-11-13:
Hmm, then what am I making of this?
```cpp
#ifdef DFU_BUILD_ENABLE
    /* Set the Vector Table(VT) base location at 0x5000 */
    NVIC_SetVectorTable(NVIC_VectTab_FLASH, 0x5000);

    USE_SYSTEM_FLAGS = 1;
#endif
```

And why does my code above not do what I'd expect it to do?

avtolstoy, 2016-11-13:
@ScruffR I assume you quoted the source code from `hal/src/core/core_hal.c`, right? As I said, at least on Photon/P1/Electron the NVIC Vector table resides in SRAM. If Core is the target platform, it is entirely possible to copy an entire vector table from flash into SRAM, replace necessary handlers and call `NVIC_SetVectorTable` again with a new address.

In order to override interrupt handlers, you would need to add something like this in `setup()`:
```c++
// See RM0033 Reference manual for the offsets
uint32_t* isrs = (uint32_t*)SCB->VTOR;
// EXTI4
isrs[0x00000068 / sizeof(uint32_t)] = (uint32_t)EXTI4_IRQHandler;
// EXTI9_5
isrs[0x0000009C / sizeof(uint32_t)] = (uint32_t)EXTI9_5_IRQHandler;
```

Also, if this is a cpp file, just in case I would declare those interrupt handlers as `extern "C"`.

dmiddlecamp, 2016-11-13:
I've definitely run into some occasions when I would have loved faster interrupts (sampling a microphone for example, and other high-frequency operations), 👍 for this from me!

ScruffR, 2016-11-13:
Doh' - true, that was Core (I thought that didn't refer to the Spark Core but the general meaning if core 😊 )
I'm not targeting the Core but Photon.

I'll give that a try

Thanks!

technobly, 2017-02-06:
This may or may not help with Photon, but I remember this working very fast for Core: https://community.particle.io/t/stopwatch-for-pinewood-derby/3486/17?u=bdub

pkourany, 2017-02-06:
@technobly, that was (2014) in the days before Photon, FreeRTOS and having the HAL!

technobly, 2017-02-06:
Yep, probably useless now!  Just going through the backlog here and trying to push things along.  @ScruffR did you ever revisit this issue and make any determinations on what was required or if you could retarget the IRQ handlers?

ScruffR, 2017-02-06:
I had been playing with it a bit when I wrote `ParticleSoftSerial` but progress was slow and I needed a solution, so I settled for lower max baudrate for the library.
But I'd still want faster interrupts for some things (e.g. dimmable RGB matrix panel, higher SoftSer baudrate).

PR #1394 (2017-12-13): Low latency (direct) interrupt handlers
<details>
  <summary><i>submission notes</i></summary>

```
**Important:** Please sanitize/remove any confidential info like usernames, passwords, org names, product names/ids, access tokens, client ids/secrets, or anything else you don't wish to share.

Please Read and Sign the Contributor License Agreement ([Info here](https://github.com/spark/firmware/blob/develop/CONTRIBUTING.md)).

You may also delete this submission notes header if you'd like. Thank you for contributing!
```
</details>

### Problem

See #1168 #1333

### Solution

This PR adds two w [... cut]
closed 2017-12-13 by m-mcgowan