particle-iot/device-os #1173: [Electron] Describe message sent twice, extra data usage.

technobly, 2016-11-15:
The describe message appears to be sent twice on the electron, consuming ~620 more bytes than necessary.

- Unconditionally sent twice on purpose due to old code
- Check to see if this is the case for the Photon as well

To test, build with #1169 and look for 2 "Describe message sent" messages or enable DEBUG_BUILD=y and observe two 620 byte messages sent in logs.

---

Completeness:
- [x] Minimum test case added
- [x] Device, system and user firmware versions stated (v0.6.0)
- [x] Particle confirmed

technobly, 2017-02-06:
@avtolstoy would you please add your thoughts to this one?  I know this was added before you did some investigation on why two describe messages might be normal.  Please provide thoughts on what the situation might have been, how it has changed, and the best practice to avoid it in the future.  Thanks!

avtolstoy, 2017-02-15:
First of all, there are two describe messages sent on Electron: system describe and application describe. They are being requested separately by the server, so all in all there should be two messages in the log telling that the DESCRIBE message is currently being sent. Commit c54ecc3bbbe03f18c2d0b0797b5a968bb1346afe introduces additional logging to know exactly which describe message is currently being sent: S - System, A - Application or both (SA), which is a case for Photon/P1/Core. The log on Electron should currently (as of 0.6.1-rc.2) present following messages:
```
0000037168 [comm.protocol] INFO: Sending S describe message
0000037168 [comm.coap] TRACE: sending message id=56e5
0000037170 [system] TRACE: send 650
....
0000037895 [comm.protocol] INFO: Sending A describe message
0000037896 [system] TRACE: send 56
...
```

Secondly, until #1140 was introduced (0.6.1-rc.1 release), we did not have proper flow control between the u.blox module and Electron MCU. Whenever sending the system DESCRIBE message, which is usually ~650 bytes, this sometimes caused u.blox module to fail to received some bytes, because it was asking Electron MCU to hold transmission, but Electron was not processing that properly. As a result DESCRIBE message was not sent to the server, which resulted in retransmissions.

m-mcgowan, 2017-10-22:
Sounds like this issue is resolved. @technobly  please close if you feel the explanation above is sufficient!

closed 2017-11-14 by technobly